rpp0/gr-lora #1: Improve synchronization

rpp0, 2016-06-27:
Since the project is still in early development, clock synchronization isn't ideal yet: this is done at the moment by looking for the slope of a downchirp and then performing a correlation of the transient with an ideal downchirp. If the argmax of the correlation result is not within a hardcoded range, the decoder will not see the frame. 

This should be changed to be more flexible and not depend on a hardcoded value. For each of the samples given in gr-lora-samples, it should work decently (captured with a HackRF). For samples captured by other devices, this might not work at all.

rpp0, 2016-08-30:
This issue is partly mitigated in the realtime decoder (Version 0.2) ad85c5e4165e3db913c3de377314cddea972f6e7. However, as constated in issue https://github.com/rpp0/gr-lora/issues/7, some LoRa radios do not send synchronization symbols. The decoder needs to be modified so that it can decode frames without them.

Wosser1sProductions, 2017-03-21:
Detecting and syncing samples have been improved greatly in #25.

closed 2017-04-24 by rpp0