rpp0/gr-lora #11: Basic Instructions

PhilColbert, 2016-11-05:
Hi,

Sorry to bother, but would it be possible if there was a basic instruction guide to getting this working, not coming from an RF background it all seems rather complicated :)

We are using LORA to talk to units whilst flying paragliders and looking to read the packets via RTL-SDR devices, is this something we could possibly do with this software ?

Thanks 

Phil.

rpp0, 2016-11-06:
Hi,

Sure, I will create an article on the wiki on how to decode messages. Your use case is very interesting! However, I would not recommend using gr-lora for this, for a number of reasons:
- gr-lora is intended for research purposes. Many aspects of how to properly decode LoRa signals are not detailed anywhere, since it is proprietary. Therefore, some LoRa messages may be corrupt or even not detected at all upon reception. 
- Decoding of high spreading factors is still slow, and I assume you would be using a high spreading factor for transmitting the LoRa packets from the paragliders to the RTL-SDR.

So all in all, I would recommend that you just use a regular, official LoRa modem with the MAC functionality disabled (so that you can communicate from node to node instead of requiring a gateway).

PhilColbert, 2016-11-06:
Thanks for the reply.

We are using  spreading factor 7 to give us the best range and its only a proof of concept at the moment, worth giving it a try.

The reason we would like to do this, is because there is already a network of ground stations running with RTL-SDR devices tracking an expensive proprietary system, our system is much much cheaper and if we can get something working it can be added into their codebase and then we can use their ground stations aswell.

Thanks for any help !

Regards

rpp0, 2016-11-24:
Hi Phil,

Apologies for the late reply; my LoRa testing device broke down and I had to replace it. You can find the RTL-SDR tutorial in the "Wiki" section of the repository. I hope it helps and if anything is unclear, please let me know.

PhilColbert, 2016-11-24:
No worries, thank you for the help, super busy at the moment but when I get
chance, will have a look.

Hope its ok to ask any questions ? :)

Phil

rpp0, 2016-11-24:
Absolutely! I'll be happy to answer any questions you have.

closed 2017-02-14 by rpp0