rpp0/gr-lora #13: AttributeError: 'module' object has no attribute 'decoder'

pcm1702, 2016-11-28:
Hello, we are opening your example (same applies for file sync and realtime example) and we are getting this error: 
Please take into account that we modified the source according to issue #12 in order to compile gr-lore.

```
Generating: '/home/guido/LoRa_GRC/gr-lora/apps/lora_receive_realtime.py'

Executing: /usr/bin/python2 -u /home/guido/LoRa_GRC/gr-lora/apps/lora_receive_realtime.py

linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_3.11.0.git-28-gc66cb1ba

Traceback (most recent call last):
  File "/home/guido/LoRa_GRC/gr-lora/apps/lora_receive_realtime.py", line 30, in <module>
    import lora
  File "/usr/local/lib/python2.7/dist-packages/lora/__init__.py", line 35, in <module>
    from lora_receiver import lora_receiver
  File "/usr/local/lib/python2.7/dist-packages/lora/lora_receiver.py", line 28, in <module>
    from lora import decoder
ImportError: cannot import name decoder
```

rpp0, 2016-11-28:
Hi, thanks for reporting. It seems that python can't find the modules. Did you install the library as specified in the instructions? If so, could you please post the output of the following commands (inside build folder):
```
$ cmake ../
$ make
$ sudo make install
$ python
>>> import sys
>>> sys.path
```

rockedscientist, 2016-11-28:
Hi! Since I encountered the same problem, I picked this issue to update. I have verified that numpy and scipy are installed, as well as liquid-dsp. You can see the results of the commands above attached in this file - [error.txt](https://github.com/rpp0/gr-lora/files/616341/error.txt).
Thanks!

rpp0, 2016-11-28:
Hello, the installation seems correct, but I noticed something strange in the error: 
```
File "/usr/local/lib/python2.7/dist-packages/lora/lora_receiver.py", line 28, in <module>
    from lora import decoder
```
This line doesn't exist anymore (see https://github.com/rpp0/gr-lora/blob/master/python/lora_receiver.py). Is it possible that you don't have the latest version?

rockedscientist, 2016-11-28:
Hi! I just did a pull about two hours ago and recompilled everything. I was able to figure out something else by looking at the closed issues. I just run through:
`sudo make uninstall`
`make clean`
Then installed swig and recompiled gr-lora. Now I don`t get the error that pcm1702 mentioned so will play around a little and see how it goes. Thanks!

rpp0, 2016-11-28:
Alright, no problem and good to hear the issue is resolved for you :). Let me know if you encounter anything else.

closed 2016-12-01 by rpp0