rpp0/gr-lora #16: Reception not working with SF > 9 and BW > 125 KHz

mgucc, 2017-01-12:
I'm doing some experiments using gr-lora on a USRP B210 and a Libelium Waspmote with SX1272 chip as a transmitter. Actually i can decode correctly (still loosing some packets as the receiver lost the synchronization with the preamble) the packets sent by the transmitter with the following parameters: Sampling Rate = 1 Msps; SF = 7, 8, 9; BW = 125 KHz; CR = 4/5, 4/6, 4/7, 4/8.
With SF = 10 the receiver recognize the preamble sometimes but never decode the packet correctly.
When i try with bandwidth 250 or 500 KHz (changing the BW hard-coded value and rebuilding), the receiver never synchronize with the preamble.
Is this behaviour a bug or an issue related to the processing of the samples?

rpp0, 2017-01-12:
Hi, this is a known issue for real time decoding. You can try to store the samples to a file first, and then use a File Source to feed the samples to gr-lora. As for the bandwidth: currently, higher bandwidths than 125 KHz are not supported yet.

rpp0, 2017-05-05:
Hi @mgucc, We've done some work on decoding higher spreading factors in real time recently, so if you want you could try again and see if the results are better.

closed 2017-08-29 by rpp0