rpp0/gr-lora #17: AttributeError: 'module' object has no attribute 'decode`

kk90, 2017-01-28:
Hello I tried To use You block in GCR but I have this error:

`Traceback (most recent call last):
  File "/home/**/top_block.py", line 110, in <module>
    main()
  File "/home/**/top_block.py", line 104, in main
    tb = top_block_cls()
  File "/home/k/top_block.py", line 79, in __init__
    self.lora_lora_receiver_0 = lora.lora_receiver(samp_rate, freq, 1, 7, 1000000)
  File "/usr/local/lib/python2.7/dist-packages/lora/lora_receiver.py", line 50, in __init__
    self.c_decoder = lora.decoder(out_samp_rate, sf)
AttributeError: 'module' object has no attribute 'decode`

All dependencies are installed, cmake with `swig` as mentioned #13 .

Below is output of my sys.path

`['', '/usr/lib/python2.7', '/usr/lib/python2.7/plat-x86_64-linux-gnu', '/usr/lib/python2.7/lib-tk', '/usr/lib/python2.7/lib-old', '/usr/lib/python2.7/lib-dynload', '/home/k/.local/lib/python2.7/site-packages', '/usr/local/lib/python2.7/dist-packages', '/usr/lib/python2.7/dist-packages', '/usr/lib/python2.7/dist-packages/PILcompat', '/usr/lib/python2.7/dist-packages/gtk-2.0', '/usr/lib/python2.7/dist-packages/ubuntu-sso-client', '/usr/lib/python2.7/dist-packages/wx-3.0-gtk2']`

rpp0, 2017-01-28:
Hi! Could you post the output of the following commands? I suspect something is wrong with the installation.
```
(inside build folder)
$ cmake ../
$ make
$ sudo make install
```

kk90, 2017-01-28:
<details> 
  <summary> <<click for command output</summary>
   k@k ~/sdr/gr-lora/build $ cmake ../
-- The CXX compiler identification is GNU 5.4.0
-- The C compiler identification is GNU 5.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Boost version: 1.58.0
-- Found the following Boost libraries:
-- filesystem
-- system
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1")
-- Checking for module 'cppunit'
-- Found cppunit, version 1.13.2
-- Found CPPUNIT: /usr/lib/x86_64-linux-gnu/libcppunit.so;dl
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)
Checking for GNU Radio Module: RUNTIME
-- Checking for module 'gnuradio-runtime'
-- Found gnuradio-runtime, version 3.7.9

    INCLUDES=/usr/include
    LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so
    -- Found GNURADIO_RUNTIME: /usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so
    GNURADIO_RUNTIME_FOUND = TRUE
    CMake Warning (dev) at /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GrTest.cmake:45 (get_target_property):
    Policy CMP0026 is not set: Disallow use of the LOCATION target property.
    Run "cmake --help-policy CMP0026" for policy details. Use the cmake_policy
    command to set the policy and suppress this warning.

The LOCATION property should not be read from target "test-lora". Use the
target name directly with add_custom_command, or use the generator
expression $<TARGET_FILE>, as appropriate.

Call Stack (most recent call first):
lib/CMakeLists.txt:81 (GR_ADD_TEST)
This warning is for project developers. Use -Wno-dev to suppress it.

--
-- Checking for module SWIG
-- Found SWIG version 3.0.8.
-- Found SWIG: /usr/bin/swig3.0
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython2.7.so (found suitable version "2.7.12", minimum required is "2")
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.12", minimum required is "2")
-- Performing Test HAVE_WNO_UNUSED_BUT_SET_VARIABLE
-- Performing Test HAVE_WNO_UNUSED_BUT_SET_VARIABLE - Success
CMake Warning (dev) at /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GrTest.cmake:45 (get_target_property):
Policy CMP0026 is not set: Disallow use of the LOCATION target property.
Run "cmake --help-policy CMP0026" for policy details. Use the cmake_policy
command to set the policy and suppress this warning.

The LOCATION property should not be read from target "gnuradio-lora". Use
the target name directly with add_custom_command, or use the generator
expression $<TARGET_FILE>, as appropriate.

Call Stack (most recent call first):
python/CMakeLists.txt:53 (GR_ADD_TEST)
This warning is for project developers. Use -Wno-dev to suppress it.

CMake Warning (dev) at /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GrTest.cmake:45 (get_target_property):
[... 77 lines cut ...]
-- Installing: /usr/local/lib/libgnuradio-lora.so
-- Installing: /usr/local/lib/python2.7/dist-packages/lora/_lora_swig.so
-- Set runtime path of "/usr/local/lib/python2.7/dist-packages/lora/_lora_swig.so" to ""
-- Installing: /usr/local/lib/python2.7/dist-packages/lora/lora_swig.py
-- Installing: /usr/local/lib/python2.7/dist-packages/lora/lora_swig.pyc
-- Installing: /usr/local/lib/python2.7/dist-packages/lora/lora_swig.pyo
-- Installing: /usr/local/include/lora/lora/swig/lora_swig.i
-- Installing: /usr/local/include/lora/lora/swig/lora_swig_doc.i
-- Installing: /usr/local/lib/python2.7/dist-packages/lora/init.py
-- Installing: /usr/local/lib/python2.7/dist-packages/lora/lora_decoder.py
-- Installing: /usr/local/lib/python2.7/dist-packages/lora/lora_receiver.py
-- Installing: /usr/local/lib/python2.7/dist-packages/lora/message_wireshark_sink.py
-- Installing: /usr/local/lib/python2.7/dist-packages/lora/init.pyc
-- Installing: /usr/local/lib/python2.7/dist-packages/lora/lora_decoder.pyc
-- Installing: /usr/local/lib/python2.7/dist-packages/lora/lora_receiver.pyc
-- Installing: /usr/local/lib/python2.7/dist-packages/lora/message_wireshark_sink.pyc
-- Installing: /usr/local/lib/python2.7/dist-packages/lora/init.pyo
-- Installing: /usr/local/lib/python2.7/dist-packages/lora/lora_decoder.pyo
-- Installing: /usr/local/lib/python2.7/dist-packages/lora/lora_receiver.pyo
-- Installing: /usr/local/lib/python2.7/dist-packages/lora/message_wireshark_sink.pyo
-- Installing: /usr/local/share/gnuradio/grc/blocks/lora_receiver.xml
-- Installing: /usr/local/share/gnuradio/grc/blocks/lora_message_file_sink.xml
-- Installing: /usr/local/share/gnuradio/grc/blocks/lora_message_wireshark_sink.xml
k@k ~/sdr/gr-lora/build $
</details>

kk90, 2017-01-31:
Hello again,
I did some debugging and I discovered that in __init__.py of lora library file at line 31.
```
try:
	# this might fail if the module is python-only
	from lora_swig import *
except ImportError as e:
	pass
```
I have an error saying 
`libgnuradio-lora.so: cannot open shared object file: No such file or directory`
I checked and file libgnuradio-lora.so is present under /usr/local/lib/libgnuradio-lora.so 
How can I fix this issue?

mgucc, 2017-02-01:
You could try just to run `sudo ldconfig` after make install, in order to link the library properly.

kk90, 2017-02-01:
That was my problem sudo ldconfig fixed my problem :smile:, maybe it would be good to add it into Readme. BTW I'm using Linux Mint 18.1

closed 2017-02-01 by kk90