rpp0/gr-lora #18: Make failed at test-lora

dienhoa, 2017-01-30:
Hello, I have this error when I try to run _make_ during installation of gr-lora on ubuntu 16.04

Scanning dependencies of target gnuradio-lora
[  4%] Building CXX object lib/CMakeFiles/gnuradio-lora.dir/decoder_impl.cc.o
[  8%] Building CXX object lib/CMakeFiles/gnuradio-lora.dir/message_file_sink_impl.cc.o
[ 12%] Linking CXX shared library libgnuradio-lora.so
[ 12%] Built target gnuradio-lora
Scanning dependencies of target test-lora
[ 16%] Building CXX object lib/CMakeFiles/test-lora.dir/test_lora.cc.o
[ 20%] Building CXX object lib/CMakeFiles/test-lora.dir/qa_lora.cc.o
[ 24%] Linking CXX executable test-lora
CMakeFiles/test-lora.dir/test_lora.cc.o: In function `main':
test_lora.cc:(.text.startup+0x21a): undefined reference to `CppUnit::XmlOutputter::XmlOutputter(CppUnit::TestResultCollector*, std::ostream&, std::string)'
test_lora.cc:(.text.startup+0x28b): undefined reference to `CppUnit::TextTestRunner::run(std::string, bool, bool, bool)'
CMakeFiles/test-lora.dir/qa_lora.cc.o: In function `qa_lora::suite()':
qa_lora.cc:(.text+0x30): undefined reference to `CppUnit::TestSuite::TestSuite(std::string)'
collect2: error: ld returned 1 exit status
lib/CMakeFiles/test-lora.dir/build.make:130: recipe for target 'lib/test-lora' failed
make[2]: *** [lib/test-lora] Error 1
CMakeFiles/Makefile2:174: recipe for target 'lib/CMakeFiles/test-lora.dir/all' failed
make[1]: *** [lib/CMakeFiles/test-lora.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

I have already install some dependencies mentions in previous issues like liquid-dsp and swig but it's still not working. I am very appreciated if someone can help me on this problem. I attach also the output of cmake in case it's needed.
[cmake output.txt](https://github.com/rpp0/gr-lora/files/739519/cmake.output.txt)

Thank you so much

P/s: I'm using BladeRF for a Receiver LoRa and I didn't see it in the list of verified Hardwares for gr-lora. Does anyone have test already with this board ?

rpp0, 2017-02-02:
Hi

Seems your compilation for a function of CppUnit, a dependency of gr-lora, is failing. Version of cppunit is okay, but the compiler you're using for C++ (4.8.5) is pretty old, so this might be the cause.

dienhoa, 2017-02-03:
Hi, 

After upgrade my C++ compiler to 5.4.1 then it works !! Thank you so much

closed 2017-02-03 by dienhoa