rpp0/gr-lora #19: Error Decoder

fcabrera23, 2017-02-12:
I have just finished installing the gr-lora without problems, however when executing examples on GNU-Radio I get the following error:

    self.c_decoder = lora.decoder(out_samp_rate, sf)
AttributeError: 'module' object has no attribute 'decode

Any idea?

rpp0, 2017-02-12:
Hi, this might help you: https://github.com/rpp0/gr-lora/wiki/Common-compile-errors

fcabrera23, 2017-02-12:
Ok thanks, now is working !! Moreover, which block can I use to get visualized the frames from Lora Receiver, im seding a "Hello" frame with my LoPy and want to see it on gnu radio. Thanks in advance!

fcabrera23, 2017-02-12:
Just manage to get the message by LoRa Message Sink. Im sending the phrase "Hello", however im getting the following output :

OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO 50 0b 01 48 65 6c ec a3 

OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO 50 0b 01 ee 03 aa 66 5d

rpp0, 2017-02-14:
The "OOO" means that your host cannot consume data fast enough, see http://files.ettus.com/manual/page_general.html#general_ounotes. Your host will miss frames and the decoded packets will be corrupt. You could try to reduce the sample rate and / or spreading factor.

closed 2017-04-24 by rpp0