rpp0/gr-lora #20: Decoding Lora packets with different spreading factors

mgucc, 2017-02-13:
Hi, I synchronized two Lora transmitters in order to have a collision of two packets with different spreading factors at the receiver (e.g. one packet with sf=7 and one packet with sf=8). I set gr-lora with both spreading factors by I can't actually decode neither of them.
Is this an issue related to real time decoding or gr-lora doesn't support this feature yet?

Thanks in advance.

rpp0, 2017-02-14:
This feature isn't supported yet, though the LoRa patent indeed claims that it should be possible. I wonder how a real LoRa device extracts the correct packet in case of a collision, it seems difficult to achieve in software.

nkaramolegos, 2018-04-04:
@mgucc hello, this is not relative to the issue but how you synchronize the two transmitters and what devices did you used? (you can send me an e-mail if you want) thank you

mgucc, 2018-04-20:
@nkaramolegos hello, I used a LoRa software transmitter that I developed with other people. I tested it with rpp0 gr-lora, with SX1272 and iC880A.

nkaramolegos, 2018-04-21:
@mgucc So you are using synchronized SDRs for TX?