rpp0/gr-lora #23: Fix 'make test'

rpp0, 2017-02-22:
The user should be able to run ```make test``` inside the build folder in order to run the unit tests, instead of executing the shell scripts in ```build/python```.

Wosser1sProductions, 2017-02-23:
At a first glance, it seems that only CppUnit tests are added to the build.
The `build/lib` directory constains a compiled version of the `lib/qa_*` tests linked together, which will be ran by typing `make test`.

The python tests might not run because they were not actually set up with the `gr_modtool` (but they were added to the `CMakeList.txt` manually).

I will look into it more thoroughly.

[Reference](http://gnuradio.org/redmine/projects/gnuradio/wiki/OutOfTreeModules)

Wosser1sProductions, 2017-03-21:
This has been fixed in #25 by editing the file paths in `qa_BasicTest_Data.xml` and `qa_BasicTest_XML.py`.

closed 2017-04-24 by rpp0