rpp0/gr-lora #29: error compiling on Ubuntu 16.04

alejoduque, 2017-05-04:
Hello, I have installed the deps and ran the commands to build until make, that fails just here:

-- Configuring done
-- Generating done
-- Build files have been written to: /home/a/Documents/gr-lora/build

a@dspstv:~/Documents/bemap/gr-lora/build$ make
[  4%] Building CXX object lib/CMakeFiles/gnuradio-lora.dir/decoder_impl.cc.o
In file included from /usr/include/c++/5/cstdint:35:0,
                 from /home/a/Documents/gr-lora/lib/utilities.h:4,
                 from /home/a/Documents/gr-lora/lib/decoder_impl.cc:32:
/usr/include/c++/5/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support must be enabled with the -std=c++11 or -std=gnu++11 compiler options.
 #error This file requires compiler and library support \
  ^
In file included from /home/a/Documents/gr-lora/lib/decoder_impl.cc:30:0:
/home/a/Documents/gr-lora/lib/decoder_impl.h:37:9: warning: scoped enums only available with -std=c++11 or -std=gnu++11
         enum class DecoderState {
         ^
/home/a/Documents/bemap/gr-lora/lib/decoder_impl.h:97:41: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
                 firdecim_crcf d_decim = nullptr;
                                         ^
/home/a/Documents/bemap/gr-lora/lib/decoder_impl.h:97:41: error: ‘nullptr’ was not declared in this scope
In file included from /home/a/Documents/bemap/gr-lora/lib/decoder_impl.cc:32:0:
/home/a/Documents/bemap/gr-lora/lib/utilities.h: In function ‘void gr::lora::print_vector(std::ostream&, const std::vector<T>&, const string&, int)’:
/home/a/Documents/bemap/gr-lora/lib/utilities.h:46:24: warning: range-based ‘for’ loops only available with -std=c++11 or -std=gnu++11
             for (T x : v)
                        ^
/home/a/Documents/bemap/gr-lora/lib/utilities.h: In function ‘void gr::lora::print_vector_raw(std::ostream&, const std::vector<T>&, int)’:
/home/a/Documents/bemap/gr-lora/lib/utilities.h:55:24: warning: range-based ‘for’ loops only available with -std=c++11 or -std=gnu++11
             for (T x : v)
                        ^
/home/a/Documents/bemap/gr-lora/lib/decoder_impl.cc: In constructor ‘gr::lora::decoder_impl::decoder_impl(float, uint8_t)’:
/home/a/Documents/bemap/gr-lora/lib/decoder_impl.cc:57:39: error: ‘gr::lora::DecoderState’ is not a class or namespace
             this->d_state = gr::lora::DecoderState::DETECT;
                                       ^
/home/a/Documents/bemap/gr-lora/lib/decoder_impl.cc: In member function ‘virtual int gr::lora::decoder_impl::work(int, gr_vector_const_void_star&, gr_vector_void_star&)’:
/home/a/Documents/bemap/gr-lora/lib/decoder_impl.cc:764:32: error: ‘gr::lora::DecoderState’ is not a class or namespace
                 case gr::lora::DecoderState::DETECT: {
                                ^
/home/a/Documents/bemap/gr-lora/lib/decoder_impl.cc:783:55: error: ‘gr::lora::DecoderState’ is not a class or namespace
                             this->d_state = gr::lora::DecoderState::SYNC;
                                                       ^
/home/a/Documents/bemap/gr-lora/lib/decoder_impl.cc:797:32: error: ‘gr::lora::DecoderState’ is not a class or namespace
                 case gr::lora::DecoderState::SYNC: {
                                ^
/home/a/Documents/bemap/gr-lora/lib/decoder_impl.cc:813:51: error: ‘gr::lora::DecoderState’ is not a class or namespace
                         this->d_state = gr::lora::DecoderState::PAUSE;
                                                   ^
/home/a/Documents/bemap/gr-lora/lib/decoder_impl.cc:818:55: error: ‘gr::lora::DecoderState’ is not a class or namespace
                             this->d_state = gr::lora::DecoderState::DETECT;
                                                       ^
/home/a/Documents/bemap/gr-lora/lib/decoder_impl.cc:829:32: error: ‘gr::lora::DecoderState’ is not a class or namespace
                 case gr::lora::DecoderState::PAUSE: {
                                ^
/home/a/Documents/bemap/gr-lora/lib/decoder_impl.cc:830:47: error: ‘gr::lora::DecoderState’ is not a class or namespace
                     this->d_state = gr::lora::DecoderState::DECODE_HEADER;
                                               ^
/home/a/Documents/bemap/gr-lora/lib/decoder_impl.cc:836:32: error: ‘gr::lora::DecoderState’ is not a class or namespace
                 case gr::lora::DecoderState::DECODE_HEADER: {
                                ^
/home/a/Documents/bemap/gr-lora/lib/decoder_impl.cc:860:51: error: ‘gr::lora::DecoderState’ is not a class or namespace
                         this->d_state = gr::lora::DecoderState::DECODE_PAYLOAD;
                                                   ^
/home/a/Documents/bemap/gr-lora/lib/decoder_impl.cc:869:32: error: ‘gr::lora::DecoderState’ is not a class or namespace
                 case gr::lora::DecoderState::DECODE_PAYLOAD: {
                                ^
/home/a/Documents/bemap/gr-lora/lib/decoder_impl.cc:887:55: error: ‘gr::lora::DecoderState’ is not a class or namespace
                             this->d_state = gr::lora::DecoderState::DETECT;
                                                       ^
/home/a/Documents/bemap/gr-lora/lib/decoder_impl.cc:902:32: error: ‘gr::lora::DecoderState’ is not a class or namespace
                 case gr::lora::DecoderState::STOP: {
                                ^
In file included from /home/a/Documents/bemap/gr-lora/lib/decoder_impl.cc:32:0:
/home/a/Documents/bemap/gr-lora/lib/utilities.h: In instantiation of ‘void gr::lora::print_vector(std::ostream&, const std::vector<T>&, const string&, int) [with T = unsigned char; std::ostream = std::basic_ostream<char>; std::__cxx11::string = std::__cxx11::basic_string<char>]’:
/home/a/Documents/bemap/gr-lora/lib/decoder_impl.cc:532:74:   required from here
/home/a/Documents/bemap/gr-lora/lib/utilities.h:47:30: error: use of local variable with automatic storage from containing function
                 out << to_bin(x, element_len_bits) << ", ";
                              ^
/home/a/Documents/bemap/gr-lora/lib/utilities.h:46:20: note: ‘unsigned char x’ declared here
             for (T x : v)
                    ^
/home/a/Documents/bemap/gr-lora/lib/utilities.h: In instantiation of ‘void gr::lora::print_vector_raw(std::ostream&, const std::vector<T>&, int) [with T = unsigned char; std::ostream = std::basic_ostream<char>]’:
/home/a/Documents/bemap/gr-lora/lib/decoder_impl.cc:594:94:   required from here
/home/a/Documents/bemap/gr-lora/lib/utilities.h:56:30: error: use of local variable with automatic storage from containing function
                 out << to_bin(x, element_len_bits);
                              ^
/home/a/Documents/bemap/gr-lora/lib/utilities.h:55:20: note: ‘unsigned char x’ declared here
             for (T x : v)
                    ^
/home/a/Documents/bemap/gr-lora/lib/decoder_impl.cc:891:41: warning: ignoring return value of ‘int system(const char*)’, declared with attribute warn_unused_result [-Wunused-result]
                             DBGR_PAUSE();
                                         ^
lib/CMakeFiles/gnuradio-lora.dir/build.make:62: recipe for target 'lib/CMakeFiles/gnuradio-lora.dir/decoder_impl.cc.o' failed
make[2]: *** [lib/CMakeFiles/gnuradio-lora.dir/decoder_impl.cc.o] Error 1
CMakeFiles/Makefile2:137: recipe for target 'lib/CMakeFiles/gnuradio-lora.dir/all' failed
make[1]: *** [lib/CMakeFiles/gnuradio-lora.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

rpp0, 2017-05-05:
@dc2mw , Thanks for helping out. I think that @alejoduque 's compiler is indeed not enabling certain c++11 features by default. Using ```gcc version 6.3.1``` should solve the problem.

alejoduque, 2017-05-06:
After adding those definitions, it compiled. thanks @rpp0

Wosser1sProductions, 2017-05-14:
The `add_definitions(-std=c++11)` tag has been added to `CMakeLists.txt` in pull request #28 as it was also needed for Travis CI to work properly.

rpp0, 2017-05-16:
Thanks for adding the definition, @Wosser1sProductions.

closed 2017-05-16 by rpp0