rpp0/gr-lora #35: AttributeError: 'module' object has no attribute 'decoder'

mmcharfi, 2017-06-19:
Hello,
I try to use your block LoRa receiver but i have the following error:
`  File "/home/mcharfi/top_block.py", line 76, in __init__
    self.lora_lora_receiver_0 = lora.lora_receiver(1e6, 868e6, 100e3, 7, 1000000, 0.01)
  File "/usr/lib/python2.7/dist-packages/lora/lora_receiver.py", line 49, in __init__
    self.c_decoder = lora.decoder(out_samp_rate, sf)
AttributeError: 'module' object has no attribute 'decoder'
OOOOOOOOOOO..`
During the installation i had some warning after use the command line:
`cmake -DCMAKE_INSTALL_PREFIX=/usr `
result: [install1.txt](https://github.com/rpp0/gr-lora/files/1085328/install1.txt)

and after the command `make`
result:[install2.txt](https://github.com/rpp0/gr-lora/files/1085338/install2.txt)

after sudo make install:
[install3.txt](https://github.com/rpp0/gr-lora/files/1085343/install3.txt).

I tried the other solution discussed in issues #13 and #17. But without success.
Please note that i used Pybombs to install gnuradio companion.
Thanks.

rpp0, 2017-06-20:
Hello there! Are you using Ubuntu? You only need to use the ```-DCMAKE_INSTALL_PREFIX=/usr``` flag if you use Arch Linux.

mmcharfi, 2017-06-22:
Hello, thanks for your reply. I 'am using Debian, but when i don't set this flag on i can't found the gr-lora module in gnuradio companion, i maybe forgot a step ?

rpp0, 2017-06-22:
You could check the following page to verify the installation was correctly performed: https://github.com/rpp0/gr-lora/wiki/Common-compile-and-runtime-errors

mmcharfi, 2017-06-29:
Thanks, I don't have this error anymore. I have uninstalled and reinstall all the stuff with pybombs. I used the way describe [here](https://www.tiny-dev.com/decoding-lora-with-hackrf-gnuradio/) that include a recipes of gr-rpp0 for pybombs.

closed 2017-06-29 by mmcharfi