rpp0/gr-lora #39: AttributeError: 'module' object has no attribute 'channelizer'

ty4tw, 2017-07-29:
After installation, I tried Testing. but I got a error messages as follows:
I checked "Common compile and runtime errors" in wiki.
all yes except site-packages, lora module is in dist-packages.

$ ./lora_receive_file.py 
Using Volk machine: avx2_64_mmx_orc
Traceback (most recent call last):
  File "./lora_receive_file.py", line 188, in <module>
    main()
  File "./lora_receive_file.py", line 182, in main
    tb = top_block_cls()
  File "./lora_receive_file.py", line 72, in __init__
    self.lora_lora_receiver_0 = lora.lora_receiver(samp_rate, 868e6, ([868.1e6]), 7, 1000000, 0.01)
  File "/usr/local/lib/python2.7/dist-packages/lora/lora_receiver.py", line 43, in __init__
    self.channelizer = lora.channelizer(in_samp_rate, out_samp_rate, center_freq, channel_list)
AttributeError: 'module' object has no attribute 'channelizer'

rpp0, 2017-07-29:
Hi, thanks for reporting! I presume that something went wrong during the installation of the library. I tried reinstalling on a clean VM and didn't encounter your issue. Could you post your build log if possible?

ty4tw, 2017-07-29:
Hi, 
Thank you for the quick response.

Is this enough for checking.
````
$ make
Scanning dependencies of target gnuradio-lora
[  8%] Building CXX object lib/CMakeFiles/gnuradio-lora.dir/decoder_impl.cc.o
[ 16%] Building CXX object lib/CMakeFiles/gnuradio-lora.dir/message_file_sink_impl.cc.o
[ 25%] Building CXX object lib/CMakeFiles/gnuradio-lora.dir/message_socket_sink_impl.cc.o
/home/tomoaki/tmp/work/gr-lora/lib/message_socket_sink_impl.cc:685:0: warning: "NDEBUG" redefined
 #define NDEBUG            /// Debug printing
 ^
<command-line>:0:0: note: this is the location of the previous definition
[ 33%] Building CXX object lib/CMakeFiles/gnuradio-lora.dir/channelizer_impl.cc.o
[ 41%] Building CXX object lib/CMakeFiles/gnuradio-lora.dir/controller_impl.cc.o
[ 50%] Linking CXX shared library libgnuradio-lora.so
[ 50%] Built target gnuradio-lora
Scanning dependencies of target test-lora
[ 58%] Building CXX object lib/CMakeFiles/test-lora.dir/test_lora.cc.o
[ 66%] Building CXX object lib/CMakeFiles/test-lora.dir/qa_lora.cc.o
[ 75%] Building CXX object lib/CMakeFiles/test-lora.dir/qa_message_socket_sink.cc.o
[ 83%] Linking CXX executable test-lora
[ 83%] Built target test-lora
Scanning dependencies of target pygen_python_2c915
[ 91%] Generating __init__.pyc, lora_decoder.pyc, lora_receiver.pyc, message_wireshark_sink.pyc
[100%] Generating __init__.pyo, lora_decoder.pyo, lora_receiver.pyo, message_wireshark_sink.pyo
[100%] Built target pygen_python_2c915
Scanning dependencies of target pygen_apps_9a6dd
[100%] Built target pygen_apps_9a6dd

$ sudo make install
[ 50%] Built target gnuradio-lora
[ 83%] Built target test-lora
[100%] Built target pygen_python_2c915
[100%] Built target pygen_apps_9a6dd
Install the project...
-- Install configuration: "Release"
-- Installing: /usr/local/lib/cmake/lora/loraConfig.cmake
-- Installing: /usr/local/include/lora/api.h
-- Installing: /usr/local/include/lora/decoder.h
-- Installing: /usr/local/include/lora/message_file_sink.h
-- Installing: /usr/local/include/lora/message_socket_sink.h
-- Installing: /usr/local/include/lora/channelizer.h
-- Installing: /usr/local/include/lora/controller.h
-- Installing: /usr/local/lib/libgnuradio-lora.so
-- Set runtime path of "/usr/local/lib/libgnuradio-lora.so" to ""
-- Installing: /usr/local/lib/python2.7/dist-packages/lora/__init__.py
-- Installing: /usr/local/lib/python2.7/dist-packages/lora/lora_decoder.py
-- Installing: /usr/local/lib/python2.7/dist-packages/lora/lora_receiver.py
-- Installing: /usr/local/lib/python2.7/dist-packages/lora/message_wireshark_sink.py
-- Installing: /usr/local/lib/python2.7/dist-packages/lora/__init__.pyc
-- Installing: /usr/local/lib/python2.7/dist-packages/lora/lora_decoder.pyc
-- Installing: /usr/local/lib/python2.7/dist-packages/lora/lora_receiver.pyc
-- Installing: /usr/local/lib/python2.7/dist-packages/lora/message_wireshark_sink.pyc
-- Installing: /usr/local/lib/python2.7/dist-packages/lora/__init__.pyo
-- Installing: /usr/local/lib/python2.7/dist-packages/lora/lora_decoder.pyo
-- Installing: /usr/local/lib/python2.7/dist-packages/lora/lora_receiver.pyo
-- Installing: /usr/local/lib/python2.7/dist-packages/lora/message_wireshark_sink.pyo
-- Installing: /usr/local/share/gnuradio/grc/blocks/lora_receiver.xml
-- Installing: /usr/local/share/gnuradio/grc/blocks/lora_message_file_sink.xml
-- Installing: /usr/local/share/gnuradio/grc/blocks/lora_message_wireshark_sink.xml
-- Installing: /usr/local/share/gnuradio/grc/blocks/lora_message_socket_sink.xml

````
If you need more info, let me know. Thank you again.

rpp0, 2017-07-31:
Looks like your Python libraries might be installed in ```/usr/lib```, while gr-lora was installed in ```/usr/local/lib```. You can try running ```cmake ../  -DCMAKE_INSTALL_PREFIX=/usr``` followed by ```make``` and ```sudo make install```.

ty4tw, 2017-07-31:
Hi,
Thank you for your quick response.

I tried cmake ../  -DCMAKE_INSTALL_PREFIX=/usr    , make, make install and sudo ldconfig.
But, nothing changed except modules were installed in /usr/lib directory.
````
[ 50%] Built target gnuradio-lora
[ 83%] Built target test-lora
[100%] Built target pygen_python_2c915
[100%] Built target pygen_apps_9a6dd
Install the project...
-- Install configuration: "Release"
-- Up-to-date: /usr/lib/cmake/lora/loraConfig.cmake
-- Up-to-date: /usr/include/lora/api.h
-- Up-to-date: /usr/include/lora/decoder.h
-- Up-to-date: /usr/include/lora/message_file_sink.h
-- Up-to-date: /usr/include/lora/message_socket_sink.h
-- Up-to-date: /usr/include/lora/channelizer.h
-- Up-to-date: /usr/include/lora/controller.h
-- Installing: /usr/lib/libgnuradio-lora.so
-- Set runtime path of "/usr/lib/libgnuradio-lora.so" to ""
-- Up-to-date: /usr/lib/python2.7/dist-packages/lora/__init__.py
-- Up-to-date: /usr/lib/python2.7/dist-packages/lora/lora_decoder.py
-- Up-to-date: /usr/lib/python2.7/dist-packages/lora/lora_receiver.py
-- Up-to-date: /usr/lib/python2.7/dist-packages/lora/message_wireshark_sink.py
-- Installing: /usr/lib/python2.7/dist-packages/lora/__init__.pyc
-- Installing: /usr/lib/python2.7/dist-packages/lora/lora_decoder.pyc
-- Installing: /usr/lib/python2.7/dist-packages/lora/lora_receiver.pyc
-- Installing: /usr/lib/python2.7/dist-packages/lora/message_wireshark_sink.pyc
-- Installing: /usr/lib/python2.7/dist-packages/lora/__init__.pyo
-- Installing: /usr/lib/python2.7/dist-packages/lora/lora_decoder.pyo
-- Installing: /usr/lib/python2.7/dist-packages/lora/lora_receiver.pyo
-- Installing: /usr/lib/python2.7/dist-packages/lora/message_wireshark_sink.pyo
-- Up-to-date: /usr/share/gnuradio/grc/blocks/lora_receiver.xml
-- Up-to-date: /usr/share/gnuradio/grc/blocks/lora_message_file_sink.xml
-- Up-to-date: /usr/share/gnuradio/grc/blocks/lora_message_wireshark_sink.xml
-- Up-to-date: /usr/share/gnuradio/grc/blocks/lora_message_socket_sink.xml

````
````
 make test
Running tests...
Test project /home/tomoaki/tmp/work/gr-lora/build
    Start 1: test_lora
1/3 Test #1: test_lora ........................   Passed    0.00 sec
    Start 2: qa_receiver
2/3 Test #2: qa_receiver ......................***Failed    0.16 sec
    Start 3: qa_BasicTest_XML
3/3 Test #3: qa_BasicTest_XML .................***Failed    0.31 sec

33% tests passed, 2 tests failed out of 3

Total Test time (real) =   0.47 sec

The following tests FAILED:
	  2 - qa_receiver (Failed)
	  3 - qa_BasicTest_XML (Failed)
Errors while running CTest
Makefile:127: ターゲット 'test' のレシピで失敗しました
make: *** [test] エラー 8
````
current PATH is
````
>>> import sys
>>> sys.path
['', '/usr/lib/python2.7/dist-packages', '/usr/lib/python2.7', '/usr/lib/python2.7/plat-x86_64-linux-gnu', '/usr/lib/python2.7/lib-tk', '/usr/lib/python2.7/lib-old', '/usr/lib/python2.7/lib-dynload', '/usr/local/lib/python2.7/dist-packages', '/usr/lib/python2.7/dist-packages/PILcompat', '/usr/lib/python2.7/dist-packages/gtk-2.0', '/usr/lib/python2.7/dist-packages/wx-3.0-gtk2']
````

rpp0, 2017-08-01:
Yes, you are right. The issue appears to be SWIG related as your log file reads ```-- Disabling SWIG because version check failed```. Could you please post the output of ```$ swig -version```?

ty4tw, 2017-08-01:
I have 2 versions  swig and swig2.0.

ty4tw, 2017-08-01:
2 swig packages were found swig and swig2.0.  but not installed.
so, did $ sudo apt install swig 
swig version is  version 3.0.8.

Then, retry  another error occcured.
````
Scanning dependencies of target lora_swig_swig_2d0df
[ 54%] Building CXX object swig/CMakeFiles/lora_swig_swig_2d0df.dir/lora_swig_swig_2d0df.cpp.o
[ 59%] Linking CXX executable lora_swig_swig_2d0df
Swig source
/home/tomoaki/tmp/work/gr-lora/swig/lora_swig.i:5: Error: Unable to find 'gnuradio.i'
swig/CMakeFiles/lora_swig_swig_2d0df.dir/build.make:105: recipe for target swig/lora_swig_swig_2d0df' failed
make[2]: *** [swig/lora_swig_swig_2d0df] Error 1
make[2]: *** delete  'swig/lora_swig_swig_2d0df'　
````
I could not find the file in work/gr-lora directry.
the file is in work/gnuradio/gnuradio-runtime/swig/ directry.

rpp0, 2017-08-02:
Not sure what the issue is here. Perhaps you don't have gnuradio-dev installed? On a clean Ubuntu (Trusty) machine, you can perform the following steps to install gr-lora:
```
    sudo apt-get update
    sudo apt-get install g++-4.8
    export CXX="g++-4.8"
    export CC="gcc-4.8"
    sudo pip install scipy
    sudo apt-get install swig
    sudo apt-get install libcppunit-dev
    sudo apt-get install fftw3 fftw-dev libfftw3-dev
    sudo apt-get install libboost1.55-all-dev
    sudo apt-get install gnuradio
    sudo apt-get install gnuradio-dev
    sudo apt-get install liblog4cpp5
    sudo apt-get install liblog4cpp5-dev
    sudo ldconfig
    git clone https://github.com/jgaeddert/liquid-dsp.git
    cd liquid-dsp/
    sudo sh ./bootstrap.sh
    sh ./configure
    make && sudo make install
    sudo ldconfig
    cd ../
    cd gr-lora
    mkdir build && cd build
    cmake ../
    make
    sudo make install
```
If you have a version older than Trusty, gr-lora is not supported unfortunately.

ty4tw, 2017-08-04:
Thank you for your support. I made it.
I uninstalled gnuradio related modules and installed again.
Every thing works fine.  I think that the installation process of gnuradion  had problems.
I did as follows:      ( add --recursive option this time)
````
git clone --recursive http://git.gnuradio.org/git/gnuradio.git
 sudo install cheetah
cd gnuradio
 mkdir build
 cd build/
 cmake ../
 make
 make test
 sudo make install
 sudo ldconfig

 cd ../../
 sudo apt-get remove gr-osmosdr
 sudo apt-get update
 sudo apt-get install gr-osmosdr
 git clone git://git.osmocom.org/gr-osmosdr
 cd gr-osmosdr/
 mkdir build
 cd build/
 cmake ../
 make
 sudo make install
 sudo ldconfig

 
cd ../../
git clone https://github.com/rpp0/gr-lora.git
cd gr-lora
mkdir build
cd build
cmake ../
make
sudo make install
make test
sudo make install
sudo ldconfig
````
I

closed 2017-08-04 by ty4tw

HTRK, 2018-10-18:
> Not sure what the issue is here. Perhaps you don't have gnuradio-dev installed? On a clean Ubuntu (Trusty) machine, you can perform the following steps to install gr-lora:
>
> ```
>     sudo apt-get [...]

@rpp0 
I followed those installation steps on a clean Trusty VM , knowing that the available gnuradio version is 3.7.2 . But when i run the python test script **lora_receive_file_nogui.py**  , i get the following error : 
[image]

And according to the discussion in  issue #14 : to avoid this error we need at least GNU Radio version 3.7.9 .