rpp0/gr-lora #43: Docker container

rpp0, 2017-08-21:
Package ```gr-lora``` in a Docker container to mitigate system-dependent issues.

zoobab, 2017-09-07:
I can spend some time now on this. Anyone else working on it?

rpp0, 2017-09-07:
Hi @zoobab, sorry, I finished packaging everything in a Docker container just now. It's in the latest release. If you're up for it, perhaps you could test whether it works for you :)?

zoobab, 2017-09-07:
Problem with using a rolling release distro like arch is that you won't be able to reproduce it in one year time.

Usually I use an ubuntu LTS to be sure nothing move (libs etc...) too much.

zoobab, 2017-09-07:
And the from image is not found:

```
$ ./docker_build_image.sh 
Sending build context to Docker daemon  12.29kB
Step 1/11 : FROM archlinux:latest
repository archlinux not found: does not exist or no pull access
Error response from daemon: No such image: rpp0/gr-lora:latest
```

rpp0, 2017-09-07:
The users shouldn't have to build the image themselves. Running `./docker_run_grlora.sh` should be sufficient to install and run. The scripts for building the image are only needed for development.

rpp0, 2017-09-15:
Several people have reported that the docker image works - at least on Linux. So I will mark this issue as closed for now.

closed 2017-09-15 by rpp0