rpp0/gr-lora #46: Nibbles seem inverted for PHDR

pierreduf, 2017-09-13:
I've decoded some real LoRaWAN packets and I've noticed that the nibbles are inverted in the PHDR (at least in the first byte which is the length of the packet)

Examples are below. The 40, 80, 00 after the first 3 bytes are OK (it's the correct LoRaWAN MType for unconfirmed uplink, Unconfirmed downlink and Join Request respectively) but the first byte seems to be the LoRa packet length but with nibbles swapped.

Of course, it's not a big deal because the rest of the packet is OK.

By the way, does someone know the meaning of the other 2 bytes  ?

```
 b1 13 0e 80 xx xx xx .. .. (length of the MAC packet is 27 or 0x1B)
 71 13 08 00 xx xx xx .. .. (length of the MAC packet is 23 or 0x17)
 71 13 08 00 xx xx xx .. .. (length of the MAC packet is 23 or 0x17)
 51 13 01 80 xx xx xx .. .. (length of the MAC packet is 21 or 0x15)
 11 13 0b 40 xx xx xx .. .. (length of the MAC packet is 17 or 0x11)
```

rpp0, 2017-09-14:
Hi, yes this is normal. Internally, the lengths are handled correctly. I will change the way they are printed to the user in the next version. 

The two other bytes contain flags (crc present / which cr) and a header CRC. Next version the CRC will also be checked :).

rpp0, 2017-09-15:
Fixed in [v0.6.2](https://github.com/rpp0/gr-lora/releases/tag/v0.6.2)!

closed 2017-09-15 by rpp0

pierreduf, 2017-09-15:
Thank you, you're unstoppable :D !

I'd like to contribute, especially to implement a Encoder/Transmitter block. How can I help ?

rpp0, 2017-09-17:
No problem, and thanks :)! It's very nice of you to help, but there are some things I need to figure out first, see #48 . I will keep you updated ;)

rpp0, 2018-02-06:
Hi @pierreduf , just wanted to let you know that we have finally started implementing the encoder :)! See https://github.com/rpp0/gr-lora/issues/48 for more information.

pierreduf, 2018-02-06:
Amazing ! Thanks a lot. I'm going to check that as soon as I can.