rpp0/gr-lora #50: Downlink messages not decoded ?

pierreduf, 2017-10-13:
Hi,

Is there any reason why the downlink messages are not decoded ? The structure is roughly the same (except there shouldn't be any CRC).

Is this intentional or is it linked to the way the dowlink packet is modulated ?

rpp0, 2017-10-15:
The modulation for downlink messages is indeed the same. My guess is that the signal strength is not sufficient for the decoder to be able to detect the frames. `gr-lora` is a lot less sensitive than real LoRa hardware. If the downlink signal is below the noise floor, it can't be detected yet. This is something I want to add in a future release, but it might take some time because it is a challenging problem.

pierreduf, 2017-10-17:
Thank you for your answer. I get that but the downlink signal was really strong (and I could indeed capture it) as my gateway was on the same desk as my device. 

Are you sure that the downlink message is exactly the same : there's no CRC (maybe that makes your code reject the message ?) and the IQ are inverted (that should change something in the way you demodulate the message ?).

pierreduf, 2017-10-23:
Sorry for the late answer. You'll find enclosed a sample of an uplink message. The characteristics are 

- Format is classical GnuRadio : **32 bits float with I&Q interleaved**
- Sample rate is **512 000 Hz**
- The center frequency is **868.3 MHz** and the message is centered
- The **uplink begins @200ms and finishes @1.6s**
- The **downlink begins @2.6s and finishes @3.67s**

The uplink message is _133120_ **8084f9980400050001659a263fc74d4047ee59** _3634_ (physical header and CRC are in italic) and downlink is **6084f99804200400017085f7aa** (I don't have the physical parts).

We clearly see that the IQ are inverted for the downlink, could that be the cause of the non detection by gr-lora ?

[image]

[lora_ul_and_dl_final.zip](https://github.com/rpp0/gr-lora/files/1407539/lora_ul_and_dl_final.zip)

HB9FUF, 2017-10-23:
Hi,

Does LoRa (i.e. PHY) itself defines a notion of uplink vs downlink?
Is it not a definition introduced in the an upper network stack-level, e.g. MAC-level like LoRaWAN?

The Semtech SX127X and SX130X chips indeed have support for "LoRa Modulation Polarity Inversion" in their SPI protocols, so that inversion looks to me as a feature of the LoRa PHY.

Looking at https://github.com/Lora-net/lora_gateway and at https://github.com/Lora-net/packet_forwarder , I see that both feature a "LoRa Modulation Polarity Invert" boolean parameter on packets to be sent by the gateway, but it leaves it to the network (i.e. the client to the packet_forwarder) to set that parameter to true/false on a per packet-level.

In LoRaWAN, in the spec 1.0.2, section "Regional Parameters" , it indeed says that normal downlink packets use "LoRa Modulation Polarity Inversion" (downlink beacon packets do not).

May I ask Pierre, which gateway you are using?  And which software you used to screenshot that LoRa signal spectrum?

pierreduf, 2017-10-24:
Hi @HB9FUF, 

I think you're right saying that LoRa PHY does not define notion of uplink vs downlink compared to other wireless technologies where uplink and downlink are on 

But, usually LoRa downlink are done with inverted IQ polarity (as it's written in the LoRaWAN specs) to have better orthogonality between DL and UL (especially where a full-duplex base station is used). Like you found, it can be done on a packet level as well and the class B beacons are not, I'm not sure why.

I captured these messages with a RTL-SDR dongle + GnuRadio (including a xlating FIR filter), the visualization and the screenshot was done with Inspectrum (https://github.com/miek/inspectrum)

rpp0, 2017-10-25:
Hi @pierreduf 

Thanks a lot, I didn't know that the polarity of the signal was inverted for the downlink channel (I only have end nodes to test). This is definitely the reason why these messages are not being decoded. Fixing it should be easy, I'll take a look at it soon ;).

tortizactility, 2018-01-03:
Hello @rpp0,

Did you take a look on the possible correction regarding the downlink IQ polarity ?

Regards,
Thibault

rpp0, 2018-01-05:
Sorry for the delay; I have been busy with some other projects recently. Anyway, the issue is fixed in https://github.com/rpp0/gr-lora/tree/dev. Tested with @pierreduf's example trace and it seems to work. I'm planning to merge in master after adding some extra features ;).

closed 2018-01-05 by rpp0