rpp0/gr-lora #51: LoRa signals not decoded correctly

habbasi3, 2017-11-01:
Hi,
I am sending 1 LoRa packet (1,2, or 3 bytes long) every second at 915 MHz (125 kHz bw) from a SX1272 LoRa end-node using the program https://os.mbed.com/users/dudmuck/code/sx127x_simple_TX_shield/. Using another SX1272 end-node as a receiver, I am able to receive and decode the message just fine. However, with my USRP B210 device using gr-lora as a receiver, I am not getting the correctly decoded payload. The payload being printed is incorrect and has incorrect number of bytes. I have tried various SF, and code rates, but it still doesn't work. 
Only under 1 specific scenario, I see the packet payload being printed out partially correctly, but only after 7 incorrectly decoded packets.

I am using 3 bytes payload (AA,CC,88). The following is my configuration in gr-lora: code rate: 4/8, SF 7, BW 125kHz, freq: 915MHz, internal sampling rate (1Msps), implicit header, and CRC (True).

After sending 18 packets (in 18 seconds), only 3 packets were partially correct as shown below. one packet gets decoded after 7 incorrectly decoded packets.
ff 88 f2
 **aa cc 88** fd c1 e2 c3
 6d 5f d7 da e1 61 42
 73 a3 65 f1 f9 e4 46
 9b be 30 b6 d0 c1 ce
 fd 92 8a f2 7d 08 62
 0f cd 27 31 b4 66 df
 f8 2f 32 58 4a 85 2a
 ff 88 f2
 **aa cc 88** fd c1 e2 c3
 6d 5f d7 da e1 61 42
 73 a3 65 f1 f9 e4 46
 9b be 30 b6 d0 c1 ce
 fd 92 8a f2 7d 08 62
 0f cd 27 31 b4 66 df
 f8 2f 32 58 4a 85 2a
 ff 88 f2
 **aa cc 88** fd c1 e2 c3

I tried every possible configuration combination but only this seemed to work partially. At sender, I am using coding rate: 0 (I have tried 1-8 but they don't work). Please let me know what am I missing. Thanks for your help.

rpp0, 2017-11-02:
The problem is implicit header mode: it is implemented differently for
different chipset vendors based on my findings. That is, some vendors
include a custom header to pass the length anyway, others don't.

In your case, it seems the transmitter is using reduced rate mode. I have a
branch that might work for the sx1272 in implicit mode, but it requires
some cleanups before I will push it. For now, I advise to use expicit
header mode, which should work in all cases.

habbasi3, 2017-11-02:
Yes, explicit header mode seems to be working. I can finally see the correctly decoded payload. Thanks a lot.

closed 2017-11-02 by habbasi3