rpp0/gr-lora #55: cannot run the example

shanggdlk, 2017-12-26:
Hi,

I successfully installed the gr-lora, but when trying to run the example, error happens as follows:

 ./lora_receive_file_nogui.py 
[?] Download test LoRa signal to decode? [y/N] y
[+] Downloading https://research.edm.uhasselt.be/probyns/lora/usrp-868.1-sf7-cr4-bw125-crc-0.sigmf-data -> ./example-trace.sigmf-data . . . . . . . . . . . . . . . . . .
[+] Downloading https://research.edm.uhasselt.be/probyns/lora/usrp-868.1-sf7-cr4-bw125-crc-0.sigmf-meta -> ./example-trace.sigmf-meta . .
[+] Configuration: 868.1 MHz, SF 7, CR 4/8, BW 125 kHz, prlen 8, crc on, implicit off
[+] Decoding. You should see a header, followed by 'deadbeef' and a CRC 5 times.
Traceback (most recent call last):
  File "./lora_receive_file_nogui.py", line 96, in <module>
    tb = lora_receive_file_nogui(data_file_path, sample_rate, capture_freq, lora_config)
  File "./lora_receive_file_nogui.py", line 30, in __init__
    self.message_socket_sink = lora.message_socket_sink('127.0.0.1', 40868, False, False)
AttributeError: 'module' object has no attribute 'message_socket_sink'

Any ideas about that? I'm running on the Ubuntu.14.04

rpp0, 2017-12-29:
Thanks for reporting. The message socket sink constructor indeed changed in the last update; I forgot to update the example accordingly.

rpp0, 2018-01-03:
Fixed in c4547e23b15702445622b20802005798dcc50a95

closed 2018-01-03 by rpp0