rpp0/gr-lora #56: Add the support of LoRaTAP and SNR calculation into the docker images

tortizactility, 2018-01-02:
Hi rpp0,

You added the support of LoRa TAP and SNR calculation into dev and master branch but unfortunately when I'm trying to enable the feature I'm getting this error : 
`message_socket_sink_impl::handle: LoRaTap is not supported yet.`

Did you have any plan to update your docker images with your last commits ?

Best Regards,
Thibault

tortizactility, 2018-01-02:
I succeed to rebuild the docker images with the following steps:

- Edit Dockerfile from `from archlinux:latest` to `from base/archlinux:latest`
- Run `docker_build_image.sh` to build the image from scratch
- Run `docker_run_grlora.sh` as usual (the script will use the latest images, so the one I created)x

Now, the LoRaTAP is correctly enabled there is no more error on my console and I'm getting decoder frame on the console. But the message sink don't seem to work from the docker. I'm doing a tcpdump and the port 40868 on lo interface and I don't see anything

rpp0, 2018-01-03:
Hi, thanks for reporting this. The issue is similar to https://github.com/rpp0/gr-lora/issues/55 : indeed I forgot to update the Docker image; I will update them as soon as possible

rpp0, 2018-01-03:
Okay, the docker images are updated now with the new message socket sink. Could you check whether the issue is resolved for you?

tortizactility, 2018-01-03:
Anyway, I have issue to use the loopback interface on docker and capture the traffic from the host.
I had to use the IP address (172.17.0.1) of docker0 interface on my host to be able to catch the traffic on wireshark.

For you information, with following steps I succeed to capture, edit and see the capture LoRa MAC packets into Wireshark : 
- tcpdump with only udp port 40868 and without ICMP packets : 
- Remove the Ethernet, IP and UDP headers (14+20+8) and set linktype to user (147):
`bittwiste -I dump.pcap -O dump_dlt_user.pcap -M 147 -D 1-42`
- Open the pcap in wireshark and in the DLT_USER parameters set the payload protocol to "lorawan" with header size equal to '14' (size of LoRa TAP). 
- Don't set the header protocol to `loratap` (for now) because there is some malformat packet error (don't know if it's due to wireshark or gr-lora)

tortizactility, 2018-01-03:
Thanks for the feedback I will double check today

rpp0, 2018-01-03:
Thanks for listing the steps to set the link type to LoRa. For your information, I used the LoRaTAP definition from https://github.com/eriknl/LoRaTap/blob/master/loratap.h, which should be the one used by Wireshark. Anyway, for now I would suggest to parse LoRaTAP packets by using a custom dissector on top of the UDP payload instead of changing the link type.

tortizactility, 2018-01-03:
If you have some tips to configure Wireshark to force LoRaTAP and LoRaWAN decoding on top of UDP i will be very interesting about :)

rpp0, 2018-01-05:
It depends on your use case: I just listen on the UDP socket and parse the datagram manually (you could use something like scapy though). Perhaps we can make another block that uses the correct linktype for Wireshark someday, but that is a different issue so I guess this one can be closed :).

closed 2018-01-05 by rpp0