rpp0/gr-lora #57: Support of Wireshark via customized Message sink block

tortizactility, 2018-01-07:
Hi @rpp0,

As discussed in another issue (closed now :) ), i'm trying to connect your project to wireshark.
Why wireshark, because since the dev version 2.5.0 the software include the support of LoRaTap and LoRaWAN.

But as LoRaWAN is not working on top of Ethernet, we need to modify the frame.
I finally succeed to make it works by following these steps : 
- Run `lora_receive_realtime` with a message sink to `docker0` IP address `172.17.0.1` instead of 127.0.0.1 (docker is not able to reach the host loopback address)
- Run the capture with tcpdump or wireshark. 
- Apply the following filter to capture `udp.dstport == 40868 && !icmp` and save the capture (or filter directly with tcpdump)
- Remove Ethernet, IP and UDP headers (total of 42 bytes) and set DLT to 147 : `bittwiste -I dump_lora_ethernet.pcap -O dump_lora_dlt_user.pcap -M 147 -D 1-42`
- In Wireshark, edit the DLT_USER settings and set it : payload protocol : lorawan, header size : 18, header protocol : loratap. I used to 18 to include LoRaTap (15 bytes) and LoRaPHY(3 bytes)

Maybe via a customized message sink block, we can change automatically remove the useless headers and set the DLT.
Please note that LoRaTAP have a dedicated DLT code (known by wireshark) : **270**
--> https://www.tcpdump.org/linktypes.html (LINKTYPE_LORATAP)

rpp0, 2018-01-08:
Thanks for the information! I agree that it would be better to use the LoRaTAP DLT instead of putting the frame on top of UDP. Since I'm currently working on some other issues, would you be up for implementing this and submitting a pull request?

tortizactility, 2018-07-23:
Hi @rpp0, 

Sorry for the delay, I didn't have time have a look on your project last months.
I know that implementing a customized Message sink with LoRa DLT will take time.

Anyway, I recheck the ressults in wireshark after using the following "Decode As" settings:
Field | Value | Type | Default | Current
------------ | ------------- | ------------- | ------------- | -------------
UDP port | 40868 | Integer, base 10 | (none) | LoRaTap
LoRa Syncword | 0x00 | Integer, base 16 | (none) | LoRaWAN

I saw that some values set as a parameter in the LoRa Receiver block can be inserted into the LoRaTap field. In my case, everything is equal to zero except the SNR

rpp0, 2018-07-24:
No problem :). The UDP approach works fine for now if you use Scapy to dissect the packets, though it is indeed not as easy to analyze in Wireshark.

Junaid-Qadir-UniGe, 2023-02-27:
Hi @tibortiz  and @rpp0, I have a problem with the gr-Lora and Wireshark since I put the port number 40868 and it converts into 40.868k. I guess due to this issue, I cannot receive packets on Wireshark. 
[image]

I kindly request you to please help me in this regard. Thanks