rpp0/gr-lora #58: Lora PHY

tortizactility, 2018-01-08:
Hi @rpp0,

During the demodulation performed by gr-lora, each time I can see 3 bytes just before the LoRa Mac Layer.
In my case, you can see the LoRaTap header and the data

```Frame 3: 36 bytes on wire (288 bits), 36 bytes captured (288 bits) on interface 0
LoRaTap header
    Header Version: 0
    Padding: 00
    Header Length: 0
    Channel
    RSSI
    Sync Word: Unknown (0x00)
Data (21 bytes)
    Data: 1221f0608ca0f40c28e0b0006ba6b1cc9406933e8c
    [Length: 21]
```
In the data, the 3 bytes related to LoRa PHY are : **12 21 f0**
**60 8c a0 f4 0c 28 e0 b0 00 6b a6 b1 cc 94 06 93 3e 8c** is the LoRaMac

Most of the time I'm getting only two possible values : **1221f0** or **0d2070**
Did you have any idea about the meaning of these values ? Is it something we should map into the LoRaTap header ?

tortizactility, 2018-01-08:
After some discussion with Wireshark, we found that the usage of the correct value of the sync word is very important in the loratap:
https://github.com/wireshark/wireshark/commit/ba92b90f8b9a1f3331ab10c2c49883ac32f577c9
Currently you are setting the value to 0x00, is it easy to decode the sync word ? Or should be set it to 0x34 (private) by default instead of 0x00?

rpp0, 2018-01-08:
> Did you have any idea about the meaning of these values ? Is it something we should map into the LoRaTap header ?

You can find the meaning of these values in https://github.com/rpp0/gr-lora/blob/master/include/lora/loraphy.h. Basically: length byte, options nibble, and CRC byte. There's also some information in the document 'AN1200.13: LoRa Modem Design Guide - SX1272/3/6/7/8'.

As for which bytes belong in LoRaTAP vs LoRaPHY: I think this is debatable, especially for the sync word. Personally, I would put all metadata (channel, rssi, implicit, reduced data rate, frequency, bandwidth etc.) in LoRaTAP, and all fields that are truly present in the LoRa frame and not part of the MAC (sync word, PHY header, payload CRC, etc.) in LoRaPHY. Note that currently, the sync word is in LoRaTAP, and LoRaTAP furthermore also misses some crucial information in my opinion. For example, there should be a boolean flag to indicate whether the frame was transmitted in implicit mode (= no PHY header), and extra property fields in LoRaTAP if this is the case (reduced rate mode, crc present in MAC, and coding rate).

> Currently you are setting the value to 0x00, is it easy to decode the sync word ? Or should be set it to 0x34 (private) by default instead of 0x00?

Yes it should be easy: an extra state would need to be added in the state machine, where the two sync words are demodulated (though the demodulation is a bit different compared to standard symbols).

eriknl, 2018-01-08:
@rpp0: When I wrote the LoRaTap v1 spec I was working with data captures by Semtech chips instead of captures directly from the air, as such it's largely based on the Semtech chip datasheets. Would it make sense to you to merge the phy and metadata into a revised LoRaTap v1.1 spec? When I started on the LoRaTap spec and got the DLT the whole idea was to set up a base for the community to improve upon.

Cheers,
Erik

tortizactility, 2018-01-08:
Glad to see that more people involved to push this project to the next level :)
I can easily share with you two captures took with gr-lora : 
1-  With LoRaTAP (DLT set to 270) + with LoRaPHY + LoRaMac : https://we.tl/Ch4FiSWZNC
2-  With LoRaTAP  (DLT set to 270) + without LoRaPHY + LoRaMac : https://we.tl/VbLCKlQ22A

rpp0, 2018-01-08:
Hi @eriknl, great to see you chiming in! I think there are two options for the PHY and metadata:

- Merge them together into LoRaTap v1.1 like you said. This seems okay to me since the PHY isn't really defined in any datasheets.
- Make a separate dissector for the LoRa PHY layer, containing the sync word, PHY header, and CRC.  This would keep all lower layer frame data in LoRaPHY and all metadata in LoRaTap, consistently with RadioTap / 802.11.

From a practical point of view, I don't think it matters much which option is chosen. My personal preference goes out to the second option, i.e. define a separate PHY layer, since at some point in the future the PHY might be extended with other fields (a nibble is currently reserved). Hence the Tap and PHY dissectors could evolve independently.

eriknl, 2018-01-09:
@rpp0 I agree that the 2nd option might be better, especially if the PHY gets extended. 

1) I would suggest we put together a v1.1 of the tap spec, I think we might want to add a new options bit field to keep track of information such as implicit mode and whether a LoRaPHY header follows the tap header. If you have any suggestions on this I'd love to hear it, especially about fields you feel are missing at the moment. The question is if it would make sense to keep a space for the sync word in the tap header in case no phy layer is present (i.e. capture from chip) as it is the only way to link it to a payload dissector. I feel a bit ambivalent about putting in a fake phy layer for this. Capturing from a chip still makes sense to me and I think it might be something that vendors would like to implement on gateways or handheld devices.
2) The current loraphy.h looks sufficient to cope with expansion, would it make sense to put a version field in there as well? We can add another byte for the syncword too.
3) Maybe we could put the tap and phy specs in one place to make things easier for other people to use the specs together.

Whatever the outcome of the above points, I would be happy to add support to Wireshark. I'll also see if I can get gr-lora working with my RTL-SDR and try to get it to capture directly to Wireshark using extcap.

rpp0, 2018-01-10:
1. Good point regarding the sync word when implicit mode is enabled. What we could do is the following:
- Add a bit flag to indicate implicit header mode. If it is disabled, a PHY header follows the LoRaTap header with the following fields: sync word (required), length (required), options (required), header CRC (required) and frame CRC (optional, depending on options field in PHY).  If it is enabled, a PHY header follows the LoRaTap header with the following fields: sync word (required) and frame CRC (optional, depending on a 'MAC CRC present bit' in LoRaTap).
- Note that in case implicit header mode is enabled, the presence of the frame CRC must be preconfigured on the LoRa node, instead of setting the CRC bit in the PHY header. Therefore, LoRaTap should provide the 'MAC CRC is present' bit flag.

I feel a bit conflicted about this, because with this approach the presence of a frame CRC is indicated by a field in LoRaTap when using implicit mode, whereas it is indicated by a field in the LoRaPHY header when using explicit mode. The same is true for the coding rate. Do you think this would be a problem?

Anyway, as for the changes required to LoRaTap I would suggest:
- Change the bandwidth to uint32_t, since there are some devices that can be configured with exotic bandwidths such as 62.5 KHz, 7.8 KHz, etc. Source: http://www.hoperf.com/upload/rf/RFM95_96_97_98W.pdf
- Add SF=6 as an additional spreading factor option (see datasheet from the first point).
- Add the following fields: 
  * IQI flag (=IQ Inversion. This could for example be used by gateways for downlink communications)
  * Implicit mode enabled bit
  * MAC CRC present bit (only visible if 'Implicit mode enabled' == true)
  * Coding rate (only visible if 'Implicit mode enabled' == true)
  * Reduced rate mode enabled bit (only visible if 'Implicit mode enabled' == true)
- In case we want to use LoRaTap in order to support frame injection as well, some extra fields could be added for the transmitter:
  * TX gain to use
  * Number of preamble symbols

2. The statement in my previous post that 'a nibble is reserved' was a bit inaccurate. It's actually only 3 bits in the header CRC field that are unused / always zero. There's no official documentation on these bits, so feel free to disregard my comment about the PHY layer possibly being extended in the future, because in hindsight I think it is unlikely that the LoRa designers will use these bits as a version field.

3. I fully agree; many thanks for your cooperation on this!

eriknl, 2018-01-13:
I'll have to take the time to read some documentation and think about what goes where before I'll get back to you :).

Off topic: Meanwhile I've added "decode as" for LoRaTap in UDP packets in Wireshark master. This way one can decode the packets sent over UDP when layer is set to LoRa TAP.
For now I'll try to make a gr block to write the messages to a fifo with a header so I can read it with extcap and pass that on to Wireshark. The nice thing is that extcap allows to add controls right into Wireshark so we can change stuff like the frequency or SF pretty much on the fly.

rpp0, 2018-01-16:
Great, thanks Erik!