rpp0/gr-lora #59: testing

nilampradhan2, 2018-02-17:
i tried to run python lora_receive_file_nogui.py file, but i got folloing error:

.. gr-lora/apps$ python lora_receive_file_nogui.py 
Traceback (most recent call last):
  File "lora_receive_file_nogui.py", line 11, in <module>
    from lora.loraconfig import LoRaConfig
ImportError: No module named loraconfig

rpp0, 2018-02-19:
Did you try this in the Docker image or on your local machine?

nilampradhan2, 2018-02-19:
now it is working.
but after i simulate lora_receive_file.grc, it shows following error:
 return _blocks_swig0.file_source_make(itemsize, filename, repeat)
RuntimeError: can't open file
how to resolve this?
thank you.

nilampradhan2, 2018-02-20:
its working!

closed 2018-03-02 by rpp0