rpp0/gr-lora #62: SX1272 - SF6, Implicit header, CR4/5, LDR enabled

end0finvention, 2018-03-15:
Hi, I'm working with an SX1272 using SF6, implicit header, CR4/5 and LDR enabled. The header portion (first two bytes) correctly decodes however the payload does not. 

I noticed that there are no test results for SF6, is this because it is known not to work correctly?

rpp0, 2018-03-16:
Hi, when a LoRa radio is configured in implicit header mode, there should be no header at all, so perhaps there is something wrong with your transmitter configuration. Since an explicit header is always transmitted in LDR mode, that would explain why the header is decoded correctly whereas the payload is not.

end0finvention, 2018-03-16:
I hadn't realised that in LDR mode there is always and explicit header so that could be the problem. The SX1272 is configured for LDR optimise on but set to implicit header mode hence my confusion. 

Would I be right in saying that SF6 decoding should work with the correct parameters set? Many thanks.

rpp0, 2018-03-16:
> I hadn't realised that in LDR mode there is always and explicit header so that could be the problem.

Sorry for the confusion, that's not what I meant. An explicit header is always sent in LDR mode, but not the other way around i.e. LDR mode does not imply that an explicit header is being transmitted.

> The SX1272 is configured for LDR optimise on but set to implicit header mode hence my confusion

Since you configured the device to use implicit header mode, there shouldn't be any header so my guess is that it's actually still in explicit header mode.

> Would I be right in saying that SF6 decoding should work with the correct parameters set? Many thanks.

It should work, but just to be sure you can try with SF 7 and see if that works for you. If SF 7 works and SF 6 does not, there might be a bug.

end0finvention, 2018-03-16:
Thanks for your reply, sorry for the crossed wires!

I have limited control over the transmitter I am investigating so sadly cannot set any other configurations to test. I do have access to the plaintext payloads which I think something is not quite decoding correctly. 

I have ordered an SX1272 dev kit which should hopefully arrive in the next few days. When it does I will run a series of tests at SF6 and let you know if I find anything.

Is it ok to keep this open until that time?

rpp0, 2018-03-16:
Sure! Thanks for taking the time to test this.

rpp0, 2018-04-11:
Yes, exactly! At least for the radios I tested, the reduced rate mode is automatically enabled for sf > 10. Is there any chance you could send me a capture of the trace you're having difficulties with? 

If each symbol value in the debug output is even, reduced rate mode is likely enabled. If this is the case for your signal, you could try adding `|| d_sf == 6` to that if-test.

end0finvention, 2018-05-24:
Hi, I believe this issue to be the same as #65. I'll close this and report any progress there.

closed 2018-05-24 by end0finvention