rpp0/gr-lora #63: Problem when SF12 is set

nkaramolegos, 2018-03-23:
I believe that the decoding using SF12 is not successful. I am testing the code using RN2483 (waspmote module from libelium) and SX1276 with an arduino. Is there any known problem?

rpp0, 2018-03-23:
SF12 should decode fine, but you can try lowering the correlation threshold
in the code and see if that gives a better result.

I'm currently investigating a better synchronization technique that doesn't
involve magic threshold values, but it will take some time to figure out.

rpp0, 2018-03-24:
Do you happen to have a `.cfile` of the signal? I could take a look at what the problem might be if you want.

nkaramolegos, 2018-03-24:
Before that, I have one overflow (O) when the packet is received which is not existing when I use the other SFs. I have to check it

ThGeoffrey, 2018-03-29:
Hi ! 

I'm interested in your answers, because I'm facing the same problem with others SF (only SF11 looks to approximately decode what I am sending with my SX1272)... I fail in decoding what I send... Strange...

ThGeoffrey, 2018-04-03:
My bad ... I realized that I have to put the "BB Gain" to 30 or 40 and the "IF gain" to 40 + the "RF Gain" to 10 to decode everything perfectly for CR8 ! That works, and this is amazing ! Now I have to modify the code to try to decode BW250 messages & try to make the other CRs work...

nkaramolegos, 2018-04-04:
@ThGeoffrey Out of curiosity, what is your SDR?

ThGeoffrey, 2018-04-06:
By the way, @rpp0 what kind of alternative do you think in order to replace the "magic thresholds for synchronization" ? PLL ???

rpp0, 2018-04-06:
The real LoRa hardware somehow uses an FFT to estimate the frequency error, but I'm not familiar with the details as this is not described in the datasheet or patent.

ThGeoffrey, 2018-04-06:
Okay... I'll try to find some piece of information though ... Thank you for all the work you did, this is just amazing ! ;)