rpp0/gr-lora #64: Why can we receive the message using SDR from node?

shaoenliu, 2018-04-09:
As we know that the node is sending message by using ABP, that means we need a confirmation to check the nwkskey, appskey and devaddr in our SDR side. How do we did that in gr-lora project?

rpp0, 2018-04-09:
Well, it's not really a confirmation; the application data will be encrypted with the AppSKey so we will see the encrypted data on the SDR side when receiving. In order to decrypt the data at the SDR side, you would have to build your own MAC layer on top of gr-lora. There are already some projects on Github that implement the LoRa MAC layer separately though.

shaoenliu, 2018-04-09:
Thanks for the answer, in my case, I have a lora node(but not in the support list) and and RTL SDR,  when I run the lora_realtime_receive.py, I can see the signal in in FFT plot at around 868MHz, but is it possible to get the real message content by SDR, if so how can we do that?

shaoenliu, 2018-04-13:
@rpp0 I am using the LoRa node that comtains iM881A and STM32L073RZ. Is the unencrypted data the data that without AppKey or?

rpp0, 2018-04-13:
If you disable LoRaWAN (the MAC layer) your data will be unencrypted, regardless of any configured AppKey. For example, with the RN2483 chip you can send the "mac pause" command to disable the MAC layer.

shaoenliu, 2018-04-16:
I am not sure it is possible or easy to disable the LoRaWAN. I would like to ask is it possible to receive the encrypted data with the known configured keys with gr-lora project?

rpp0, 2018-04-16:
Yes, it is possible to receive the encrypted data.

shaoenliu, 2018-04-17:
And in the "Message socket sink"  , which layer should I choose. I tried the different layers, but in the wireshark side when I input the "udp.dstport == 40868 && !icmp" it show syntax error, and what could be the reason?

rpp0, 2018-04-18:
It sounds like you actually want to decrypt the encrypted frames and not just receive them. Like I said before, gr-lora can't do that, and you have two options:

- Build your own MAC layer on top of gr-lora or use a project that implements the LoRa MAC layer (there are a few out there on Github). gr-lora sends the received data over a UDP socket, so you can use scapy, libpcap, etc.
- Disable the MAC layer on your LoRa node.

`gr-lora` is not intended to be a full-stack LoRa solution. It's just a decoder for the PHY layer signals. What happens on top of the PHY must be handled by other applications.

rpp0, 2018-04-22:
Alright, good luck with your project! Since the issue is not related to `gr-lora` I will close this one. Feel free to open a new issue if you encounter any problems with receiving the physical-layer payload.

closed 2018-04-22 by rpp0