rpp0/gr-lora #66: Use of channel list in receiver block

ElyKar, 2018-05-22:
Hi, thanks for your amazing work on LoRa demod & decoding, I really enjoyed your paper :)
I wanted to toy a bit with gr-lora, get it to work and understand a bit more in depth what happens.

So I got it to work without much problem, with RN2483 as LoRa-MAC emitter and a classic RTL SDR as receiver.
You mention in your paper (A Multi-Channel Software Decoder for the LoRa Modulation Scheme) that this software supports decoding multiple channels at the same time. I thought this was implemented with the "channel list" of the Lora receiver block, however when there is more than one element in the list (tried with [868.1e6,868.3e6,868.5e6]) Gnuradio can't execute the program and fails with the error : `RuntimeError: In hierarchical block channelizer, output 1 is not connected internally`.

I am running Arch Linux (kernel 4.16.8-1), GNU radio version 3.7.11, gr-lora is compiled from source from this very repo (commit 60e94ed).
I tried looking a bit at the implementation, but I have yet to catch up on Gnuradio's internal API which I know nothing about.

rpp0, 2018-05-22:
Thanks, glad you liked the paper :)! The multiple channel mode can be enabled by setting the option "Disable Channelization" to "Yes" in the "LoRa Receiver" block. What this will do is remove the low pass filter so that the receiver no longer sees only 125 KHz of the signal, but the entire bandwidth of the SDR. Note that this will degrade the signal quality a bit though.

Listing multiple channels in the "Channel List" is something that I indeed planned to implement but haven't finished yet, which is why you get the error (so, my bad). The difference with disabling channelization is that in this case, the receiver should create 3 separate low pass filters and pass each of the resulting streams to the decoder. However, this approach is really slow, so I didn't pay much attention to the feature. As a temporary workaround you could just create 3 separate "LoRa Receiver" blocks with different center frequencies and channel lists of size 1. This would practically give the same result.

ElyKar, 2018-05-24:
Thanks for your answer !
Duplicating the receiver works perfectly using a source file. Unfortunately, my computer is not beefy enough to support demodulating 3 channels in real time, but it's not a big issue.

rpp0, 2018-05-24:
Great! I'll keep this issue open until the channel list feature is fixed.

ElyKar, 2018-05-24:
Great, as you wish.