rpp0/gr-lora #76: Problem with coding rate (4,7)

wphilips, 2018-09-04:
I have been investigating the internals of gr-lora and I found something which may 
point to a bug for coding rate different from (4,8). 

The way I test is as follows;
-I have a working configuration where packets are transmitted  with SF=8 BW=125kHz  using 
a hope rf module, coding rate either 8 or 7 the message transmitted is 15 bytes. Either The string ?elloHelloHello where ? runs over the range A...Z, or 15 zeros

Both for coding rate 8 and 7, the message is decoded successfully by your code. 
However I am currently interested in the bit errors before forward error correction.
To summarize: with coding rate 8, these bit errors are almost non existent. However,
if I switch to coding rate 7, many errors start to appear in fixed (but data dependent) positions.
For instance, when transmitting 15 zeros the following dewhitened bytes have errors:
 1 3 5 6 7 9 10 15 19 20 26 28 29 34 35 36 37

Most often the error is in the MSB.  For example
print /x  d_words_dewhitened 
$4 = std::vector of length 41, capacity 64 = {0x0, 0x80, 0x0, 0x80, 0x0, 0x80, 0x80, 0x80, 0x0, 0x80, 0x80, 0x0, 0x0, 0x0, 0x0, 0x80, 0x0, 0x0, 0x0, 0x80, 0x80, 0x0, 0x0, 0x0, 0x0, 0x0, 0x80, 0x0, 0x80, 0x80, 0x87, 0x2d, 0xcc, 0x78, 0xad, 0x80, 0x9e, 0x9e, 0x87, 0xaa, 0xff}
This seems to be typical (errors in the MSB only)

In case I send the ?Hello... string, 
the list of erroneous positions starts with  1 6 8 9 10 11 13  if  the '?' symbol is P or above 
and with 6 8 ... otherwise.

My initial guess is that the dewhitening sequence may  need to be changed in this case.
Do you think this could be the reason? Do you have any suggestions (apart from correcting the 
dewhitening sequence and seeing what happens. I will test that sometime)?

Here are some more details:
1. I have also been experimenting with an a fft based demodulator (not gr-lora but something I cooked up myself). It does not work fully yet as it sometimes cannot synchronize properly, but when it does sycnhronize it seems to be quite accurate. The demodulated symbols produced are mostly the same  as in gr-lora, but there are some exceptions. In gr-lora I sometimes see additional bit errors, but not frequently. This leads me to believe the problem is not in the demodulator. Another reason is that  the errors are too predictable.

FYI: I need to investigate further, but in one case I noticed that a "symbol 0 chirp" (one which has no downwards jump") seemed to be mis-estimated as a symbol which was much different  in gr lora,
whereas my code produced the correct 0. I remember seeing a graph that lead me to believe the problem
was in gr-lora. If I can reproduce, I will create a ticket.

2. the code modified as follows for testing. It is as simple as it is messy. It may be  useful feature
to add such fec_encoding code to gr-lora simply for measuring the bit error rate.

`                 //hamming_decode_soft(is_header);
									uint32_t n = ceil(d_words_dewhitened.size() * 4.0f / (4.0f + d_phdr.cr)); // is_header:3 ; !is_header: 13
									uint8_t decoded[n]; //n=24

									std::vector<uint8_t> test_encoded;
									test_encoded.resize(ceil(n*(d_phdr.cr+4)/4)); //size=41 
									
									fec_decode(d_h48_fec, n, &d_words_dewhitened[0], decoded);
									fec_encode(d_h48_fec, n, decoded, &test_encoded[0]);
									{
										int i;
										int num_errors=0;
										for (i=0; i< d_words_dewhitened.size(); ++i) {
											if(d_words_dewhitened[i]!= test_encoded[i])
												num_errors++;
										}
										if (num_errors>0) {
											printf("\nis_header=%d ERRORS=%d\n", is_header, num_errors);
											printf("errpos: ");
											for (i=0; i< d_words_dewhitened.size(); ++i) {
												if(d_words_dewhitened[i]!= test_encoded[i])
													printf("%d ",i);
											}
											printf("\n");
										}
									}
									
	`

wphilips, 2018-09-05:
On reflection: this may be due to the way fec 4/7 is implemented. 
The code runs 4/8 instead. Probably one of the computed parity bits should
be set to zero. Wikipedia suggests that it is the lsb, but this may be misleading.

wphilips, 2018-09-05:
It seems to be more complicated than that. Either the way to obtain a (4,7) code from (4,8) is
more complicated than discarding a parity bit (it shouldn't be) or something is going wrong 
in the decoder, but in such a way that the fec_decoder is still able to correct.

Or am I missing something?

wphilips, 2018-09-08:
After further debugging, I finally understand how the forward error coding is implemented.
The conclusion is that the code is correct, also for coderate (4,7) 
but I think it is useful to add some additional information below for others interested.

Basically, for any coding rate cr, the interleaver packs cr received bits into a byte, setting the remaining bits 
to zero. E.g., for cr=7, the resulting bytes contain 7 received bits and 1 zero bit.
These zeroed bits can potentially be moved by the deshuffling process, but they remain 
in the msb positions  for coderates (4,7) and (4,6).
Dewhitening changes these bits to 0 or 1 depending on the  datat nibble index. The extra bit depends
only on the nibble index in the sent data.

The code then always performs forward error correction with hamming code (4,8)
irrespective of the code rate. This is quite confusing, but it works because hamming (4,8) is basically equivalent to applying (4,7) and adding an extra parity bit. It is also more efficient than packing the 
bits into fewer bytes and then calling fec_decode with parameter (4,7).

For coderate (4.7) data, the fec_decode call with parameter (4,8) basically operates on bytes 
which each contain 7 received bits and one 'bad' bit which depends only on the index of the byte. 
This bit is treated as a bit error by fec_decode, which is (usually) able to correct this. This decoding
solution is probably optimal (not 100% sure).

In case you are wondering why I am doing all of this. My goal is to detect bit errors prior to forward error correction. A proper way to find these errors  is therefore: 
1. re-encode the coded message with hamming (4,8)
2. compare the resulting encoded message with the dewhitened words. Check for any 
difference in bits, but ignore the most significant bit for (4,7) or the two most significant bits for (4,6).
For (4,8) no bits have to be ignored.

What is the use?
1.  the number of correctable bit errors is an indication of the quality of demodulation and of the 
signal snr. I am mostly interested in the quality of demodulation.

2. I wish to obtain some examples of real received messages for which the demodulated  symbols (bin_idx values in the code) are correct with high probability. This helps to understand any errors in the demodulation
process at high snr.

closed 2018-09-08 by wphilips

rpp0, 2018-09-17:
Hi, apologies for the late reply and thanks for your detailed analysis! It would indeed be a good idea to detect the number of bit errors and report this to the user somehow. Liquid-sdr does not support this out of the box unfortunately, but I like your idea of re-encoding the message and comparing with the dewhitened codewords. Personally I did this by implementing the Hamming coding manually, but your approach is better. Although I suppose this would only be useful for debugging (i.e. not in production), since we have to do an extra encoding operation for each nibble. 

Did you have any further success with your FFT-based demodulator? The hardware has an implementation that is capable of receiving below the noise floor, but I don't know how it works exactly.

wphilips, 2018-09-17:
My fft-based demodulator is starting to work: for many test signals I record, it starts to produce
the same results as gr-lora, even on very long sequences (254 bytes). However it still has a few 
problems: sometimes it synchronizes poorly on some signals. For long signals, I see some 
frequency or timing drift occur.  With ldr=True it decodes correctly, but with ldr=False probably not.
This can be fixed I think.

wphilips, 2018-09-17:
I first need to investigate if the drift is frequency or time or both. Your approach is feasible, but doubles
the computation time, which is already high for fft. I have a potentially much faster one, which usually 
but not always works well on the preamble. It needs some work and it may need extension for 
data symbols. Work in progress.