rpp0/gr-lora #82: You can verify whether the LoRa modem is working correctly by transmitting a message at 868.1 MHz.

wwwzrb, 2019-01-15:
"You can verify whether the LoRa modem is working correctly by transmitting a message at 868.1 MHz. "
For this part, I don't understand why transmitting messages at 868.1MHz since the center carrier frequency is 868MHz. Besides, I'm also confused about this "A LoRa signal should appear centered at 868.1 MHz in the FFT plot during the transmission!"

rpp0, 2019-01-15:
The SDR center frequency is indeed tuned to 868 MHz because we want to mitigate leakage of the LO, which manifests as a DC bias (i.e. a peak at 0 Hz) in the spectrum. You can observe this effect in GQRX or any other software spectrum analyser. In gr-lora, the channelizer block then digitally shifts the signal so that it is centered at 868.1 as you would expect. This technique is called "offset tuning".

wwwzrb, 2019-01-15:
Thank you very much for your reply! I'm still confused about why the center frequency is not identical in the transmitter and receiver. In the LoRa node as a transmitter, the carrier center frequency is 868MHz. The capture_freq of SDR is also set at 868MHz. So Why "A LoRa signal should appear centered at 868.1 MHz in the FFT plot during the transmission!". If USRP and GNURadio are used to analyze the FFT plot of the received signal, will a similar phenomenon appear that the center frequency has a difference of 0.1 MHz even though the devices have no frequency offset?

wwwzrb, 2019-01-15:
I know what you meant now. For LoRa operating at the 868MHz band, the center frequency of ch0 is actual 868.1MHz. Since frequency offset usually exists in non-perfect devices, you tune the receiver frequency to the center frequency with specific offset. This technology is commonly used and I didn't the center frequency at 868MHz is 868.1MHz, 868.3MHz, ..., before.

rpp0, 2019-01-16:
> I know what you meant now. For LoRa operating at the 868MHz band, the center frequency of ch0 is actual 868.1MHz. Since frequency offset usually exists in non-perfect devices, you tune the receiver frequency to the center frequency with specific offset. This technology is commonly used and I didn't know the center frequency at 868MHz is 868.1MHz, 868.3MHz, ..., before.

The fact that the SDR is tuned to 868 MHz instead of 868.1 MHz has nothing to do with the frequency offset of the LoRa devices. A frequency offset of 100 KHz, assuming stationary devices, would be quite large and as far as I know doesn't occur in practice in the case of LoRa. Like I said in my first comment, the signal received by the SDR is digitally shifted to 868.1 MHz, so the center frequencies do match.

wwwzrb, 2019-01-17:
My explanation is not very clear. I mean the frequency offset usually exists in the non-perfect SDR devices so we have to manually shift the center frequency of the received signal.

For the channelizer, I'm still confused. Do you mean that the center frequency in the transmitter is 868.1 MHz, and the signal received by the SDR is manually shifted to 868.1 MHz, as a result, the center frequencies are matched!