rpp0/gr-lora #83: Detailed ususage and parameter setting!

wwwzrb, 2019-01-15:
Thanks a lot to your contribution to the open-source gr-lora! Since this repository has been updated for a number of versions, some new features have been supported, e.g., different spreading factors, implicit and explicit header mode, etc. However, the configuration in the wiki is still the old one. Can you introduce detailed parameter settings for the new version LoRa receiver block, e.g., the meaning of "conjugate downlink signal" and "decimation"? Besides, the data set from gr-lora-samples also has no specific introductions, e.g., header mode, disable channelization and drift correction or not. I don't know the detailed setting for the updated blocks to receive these data set.

rpp0, 2019-01-16:
You're right, the documentation should be updated but I'm struggling to find the time to work on this project, though I would like to.

About your two questions:
1. Do you mean that you want to see the LoRaTAP metadata in Wireshark? This is not supported yet. Currently, the LoRaTAP header is embedded in a UDP transport, which is why Wireshark just sees it as data.
2. Assuming you mean the RN2483: nope it doesn't support implicit header mode. You can send any command you like using `lc.serial_sr(<command>, [<args>])`

wwwzrb, 2019-01-17:
okay, thanks a lot!

Do you know any devices that can support implicit mode?