rpp0/gr-lora #84: Some problem about decoder_impl::fine_sync!

wwwzrb, 2019-01-18:
### `        void decoder_impl::fine_sync(const gr_complex* in_samples, uint32_t bin_idx, int32_t search_space) {
            int32_t shift_ref = (bin_idx+1) * d_decim_factor;
            float samples_ifreq[d_samples_per_symbol];
            float max_correlation = 0.0f;
            int32_t lag = 0;

            instantaneous_frequency(in_samples, samples_ifreq, d_samples_per_symbol);

            for(int32_t i = -search_space+1; i < search_space; i++) {
                //float c = cross_correlate_fast(in_samples, &d_upchirp_v[shift_ref+i+d_samples_per_symbol], d_samples_per_symbol);
               float** c = cross_correlate_ifreq_fast(samples_ifreq, &d_upchirp_ifreq_v[shift_ref+i], d_samples_per_symbol);
                // wz: the begin position of d_upchirp_ifreq_v is shift_ref or shift_ref+d_samples_per_symbol!
                if(c > max_correlation) {
                     max_correlation = c;
                     lag = i;
                 }
            }

            #ifdef DEBUG
                //d_debug << "FINE: " << -lag << std::endl;
            #endif

            d_fine_sync = -lag;

            //if(abs(d_fine_sync) >= d_decim_factor / 2)
            //    d_fine_sync = 0;
            //d_fine_sync = 0;
        }`
###                float** c = cross_correlate_ifreq_fast(samples_ifreq, &d_upchirp_ifreq_v[shift_ref+i+d_samples_per_symbol], d_samples_per_symbol);

In the fine_sync function, input samples "samples_ifreq" are correlated with "d_upchirp_ifreq_v". Should the beginning position of "d_upchirp_ifreq_v" be "shift_ref+i"? Since "shift_ref" is equal to "d_samples_per_symbol", if the correlation begins from "shift_ref+i+d_samples_per_symbol", this will be out bound when i is larger that 0 as the length of "d_upchirp_ifreq_v" is "3*d_samples_per_symbol".

rpp0, 2019-01-18:
I think you are right. It seems the code was changed to bin_idx+1 at some point without updating the d_upchirp_ifreq_v buffer, which will cause a buffer overread if bin_idx is 127. The d_upchirp_ifreq_v buffer should contain 4 chirps to fix. Since you found this issue, feel free to change the code and submit a pull request.

wwwzrb, 2019-01-21:
I think three chirps is fairly enough for "d_upchirp_ifreq_v" buffer and overread will not happen is the correlation begins from the second chirp in the buffer! Since I have changed so much of the "origin: master" code, I cannot submit a pull request with the changed code. This problem also will not affect a lot and it runs right in the local samples test cases.