rpp0/gr-lora #97: LimeSDR compatibility?

matiassequeira, 2019-08-22:
Hi, 

I was wondering if it's possible to use a LimeSDR with gr-lora? Also, if it's possible to listen to both uplink and downlink traffic simultaneously (I know that it may depend on the HW)?

Best

rpp0, 2019-09-06:
I'm not familiar with LimeSDR, but if it's compatible with GNU Radio then it should work. However, I know MyriadRF developed a LoRa modem specifically for LimeSDR, so you might want to check that out as well: https://github.com/myriadrf/LoRa-SDR and https://myriadrf.org/news/lora-modem-limesdr/

mustard123, 2019-09-15:
I use it with LimeSDR (with limesdr plugin for gnuradio) and it works fine

matiassequeira, 2019-09-17:
> I use it with LimeSDR (with limesdr plugin for gnuradio) and it works fine

Hi mustard123! How did you installed everything? Through Docker? If you'd like to provide few directions it'd be awesome 😄.

mustard123, 2019-10-03:
Check this guide: https://wiki.myriadrf.org/Gr-limesdr_Plugin_for_GNURadio

riatankarsahu, 2021-02-08:
Hello, 
Yeah it works perfectly with LimeSDR. I made a big project with it. And it's more perfect than others. Thank you @rpp0 
-- Ritam

ph03n1xAu, 2021-02-25:
Hi Ritam @riatankarsahu,

Glad to know that it's working with LimeSDR. I tried with arduino-lora shield and also the lora sample files available in this repo..I'm getting nothing in the decode socket.. Could you please share a grc so that I can check the lime source config.

Thanks..

riatankarsahu, 2021-02-25:
Hey @ph03n1xAu ,
Good to see you again! please find the attachment! And also could you please tell me about your configurations?
-- Ritam

[LoRa_Test_3.zip](https://github.com/rpp0/gr-lora/files/6041193/LoRa_Test_3.zip)

ph03n1xAu, 2021-02-26:
Hey Ritam @riatankarsahu 

Thanks for sharing mate..It turns out my vm was not happy with the installation process that I did last time, So, I reinstalled everything from scratch yesterday evening.

Now it's working perfectly.. Wohoo :-D!! 

I was able to decode messages from Arduino-lora with LimeSDR Mini
[image]

Next up, Transmitter Experiment with LimeSDR

Hi @matiassequeira,

If you are still having problem, let me know..I just freshly build everything in both ubuntu 18.04 with gr3.7 and ubuntu 20 with gr3.8..I hope I'll be able to share some installation log with you to make it work.

Thanks to @rpp0 and other contributors for this awesome projects :-) 👍

riatankarsahu, 2021-02-26:
Hey @ph03n1xAu 
Glad to know that you made it work!! Congrats mate!! gr3.7, did it work properly? Also if you can implement transmitter, please share it with me. I need to work on the transmitter either. Thank you!

ph03n1xAu, 2021-02-28:
> Hey @ph03n1xAu
> Glad to know that you made it work!! Congrats mate!! gr3.7, did it work properly? Also if you can implement transmitter, please share it with me. I need to work on the transmitter either. Thank you!

Hi @riatankarsahu,

Yeah, gr3.7 branch worked perfectly under ubuntu 18.04 with LimeSDR mini.
For sure, I'll let you know if any success with Transmitter. If I can do it, anybody in this planet will be able to do that :-)

Cheers!