rpp0/gr-lora #101: Decode package generated from encoder

mustard123, 2019-09-24:
Hello, yes there's an experimental block in the "encoder" branch of this repository that can transmit a (currently hardcoded) LoRa frame that can be received by real LoRa hardware. However, there are some artifacts at the symbol boundaries. The real hardware can compensate for those apparently, but the software decoder cannot.

_Originally posted by @rpp0 in https://github.com/rpp0/gr-lora/issues/48#issuecomment-402489647_

What would need to be done to resolve the aforementioned artifact issues at the symbol boundaries? Also my gateway says "CRC BAD" on receiving packages from the encoder branch. How do you calcuate the CRC on the hardcoded frame?

rpp0, 2019-09-24:
I don't know. You could use an existing packet sent from a LoRa node or disable CRC checking on the gateway.

Backup-Toaster, 2020-04-08:
A test packet that I have used previously, that was copied from a SX1262 dev board, with valid CRC is:

`char test_pkt[] = "\x00\x00\x12\x00\x00\xa1\xbc\x33\x01\x07\x00\x00\x00\x00\x00\x12\x91\x60\x00\x00\x00\x00\x01\x50\x45\x52\x08\x09\x0a\x0b\x0c\x0d\x0e\x0f\x10\x11\xd4\xcc";`

Per the artefacts at the symbol boundaries, these are related to the way the code is packaging codewords into interleaved blocks. See my discussion [here](https://github.com/rpp0/gr-lora/issues/109#issuecomment-610782953).