rpp0/gr-lora #107: Off by 1 after transmitting

jason955, 2019-10-22:
Thanks very much for all your hard work here! I'm currently trying to transmit the fixed test packet provided for the encoder branch and have noticed a bug. After all the encoding we begin to send our buffer of symbol values out over the air to our receiver. If these values are (10,20,30,40) when they are demodulated by the max_gradient function and then stored in bin_idx they are reported as (9, 19, 29, 39). They are all off by one and regardless of length of transmission this pattern holds true. Do you have any idea what could be going on here?

Note: when I test this with an implicit header it is (luckily) decoded into the correct values. This is because of the rounding function triggered by `reduced_rate` being automatically set for implicit header mode. When the implicit mode transmission is demoded it still reports off by 1 errors but the division and then rounding in the std:lround line overrules the problem giving us the correct number after grey decoding occurs. Also, adding bin_idx++ when in explicit mode fixes this problem but then calls all other tests to fail.

TL;DR

Explicit Mode
Transmission: (10, 20, 30, 40)   
bin_idx val from receiver: (9, 19, 29, 39)
bin_idx after grey: (completely wrong values)

Implicit Mode
Transmission: (10, 20, 30, 40)   
bin_idx val from receiver: (9, 19, 29, 39)
bin_idx after grey: (10, 20, 30, 40)

Why off by 1

rpp0, 2019-10-23:
Hi, thanks for reaching out. To be honest, I'm not sure about the cause of this. It could be the encoder doing things differently than the LoRa hardware or it could be due to the gradient-based decoding. The real LoRa hardware seemed at least to be able to decode the packet if I remember correctly. So, one way to fix this issue would be to implement FFT-based decoding, which I currently don't have the time for unfortunately.

jason955, 2019-10-23:
Do you have any documentation on FFT-based decoding?

jason955, 2019-10-23:
Also could you describe your "real LoRa hardware" I currently am using a rhf0m301 gateway and am unable to receive your explicit test packet.

rpp0, 2019-10-24:
As far as I know, there is no official documentation. But this paper came out this year: https://hal-cea.archives-ouvertes.fr/cea-02280910/document. 

I tested on a RN2483, which could receive the packet. However, you need to disable CRC checking because the example packet doesn't include a valid CRC.

Amalinda, 2019-10-24:
Thanks for sharing the paper!

jason955, 2019-10-24:
Nevermind I changed 1 setting and it works now thanks!

Backup-Toaster, 2020-04-08:
I will comment some findings here as I had the same problem and wished to share my solution in case it helps another. I have a pull request I am working on which should address this issue too.

Short answer: Add 1 modulo 2^SF to fix this error:

```C++
for(uint8_t symbol = 0; symbol < 4+cr; symbol++) {
    symbols[symbol] = gray_decode(symbols[symbol]);
    if(reduced_rate)
        symbols[symbol] <<= 2;
    symbols[symbol] += 1; // Add one.
    symbols[symbol] %= 1u << (reduced_rate ? sf + 2 : sf); // Modulo 2^SF
    std::cout << "(" << (int)symbols[symbol] - 1 << "; " << (int)symbols[symbol] << "), ";
}
```

I agree with above that the gradient determination implementation of, what is likely, an FFT based decoding is causing this. Simply, the decoder is returning `max_index = i+1;` (line 486 of decoder_impl.cc on master branch), whereas the encoder is creating the chirps without that 'plus 1'. The implementation in the decoder is ok though, because it works with real LoRa hardware.

I hope that helps.