rpp0/gr-lora #108: No module named 'gnuradio', Raspberry Pi 4 with Raspbian

junqing-zhang, 2019-11-15:
Hello, I am using a Raspberry Pi 4 with the latest Raspbian. I installed gnuradio, GNU Radio Companion 3.7.13.4. 

When I ran the test "./lora_receive_file_nogui.py", I got the error below.
Traceback (most recent call last):
  File "./lora_receive_file_nogui.py", line 5, in <module>
    from gnuradio import blocks
ModuleNotFoundError: No module named 'gnuradio'

Has anyone had same issue before? Thanks.

mustard123, 2019-11-28:
Try running sudo ldconfig after installing so the linker gets informed about the newly installed modules

Hudson7777, 2020-09-25:
> Hello, I am using a Raspberry Pi 4 with the latest Raspbian. I installed gnuradio, GNU Radio Companion 3.7.13.4.
> 
> When I ran the test "./lora_receive_file_nogui.py", I got the error below.
> Traceback (most recent call last):
> File "./lora_receive_file_nogui.py", line 5, in
> from gnuradio import blocks
> ModuleNotFoundError: No module named 'gnuradio'
> 
> Has anyone had same issue before? Thanks.

Hi ！Have you already solved your problem? I encountered same question recently and haven't find a way to solve it.

riatankarsahu, 2021-02-08:
Hello @Hudson7777,
Was it related to PythonPath? I am going to work with same thing soon. That's why I wanted to know. Thank you in advance. 
-- Ritam