rpp0/gr-lora #113: Docker container use question

nickterry, 2020-03-29:
I created a VM running Kubuntu 18.04.4, installed docker and the LoRa decoder docker container.
Running ./lora_receive_file_nogui.py produces the expected results from the downloaded file.
I would now like to try decoding off air using my NESDR Mini 2+ dongle:

The Linux host sees the dongle:
[  759.013692] usb 1-7: Product: RTL2838UHIDIR
[  759.013692] usb 1-7: Manufacturer: Realtek
[  759.013693] usb 1-7: SerialNumber: 00000001
[  759.323729] usb 1-7: dvb_usb_v2: found a 'Realtek RTL2832U reference design' in warm state

However, I am unsure how to configure the software in the container to use this dongle. Attempts to run any of the scripts results in an error (e.g.): 

[root@84dcc5275ba8 apps]# sudo ./lora_receive_file.py       
/usr/lib/python2.7/site-packages/wx-3.0-gtk3/wx/_core.py:16629: UserWarning: wxPython/wxWidgets release number mismatch
  warnings.warn("wxPython/wxWidgets release number mismatch")
Unable to access the X Display, is $DISPLAY set properly?

Any help to get this running would be much appreciated.
Thanks

rpp0, 2020-03-30:
Hi! To run GUI applications in Docker, you need to pass some extra parameters to your docker run command. See http://fabiorehm.com/blog/2014/09/11/running-gui-apps-with-docker/ for an example.

nickterry, 2020-04-01:
Thanks for following up. Based on that article and some further research I tried the simplest option of "$ xhost +" in the the host machine before starting the container. Now when I run "$ sudo ./lora_receive_realtime.py" I make more progress. Some of the UI is presented and I can see that there is activity on the SDR dongle. BUT, I get a lot of "WARNING" and "CRITICAL" messages while the UI is booting and I suspect lots of the UI are just missing e.g.:

[root@kbuntu-vm apps]# sudo ./lora_receive_realtime.py
/usr/lib/python2.7/site-packages/wx-3.0-gtk3/wx/_core.py:16629: UserWarning: wxPython/wxWidgets release number mismatch
  warnings.warn("wxPython/wxWidgets release number mismatch")
linux; GNU C++ version 7.2.1 20171224; Boost_106600; UHD_003.010.002.000-0-unknown

libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.11
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 1000000.026491 Hz
Bits (nominal) per symbol:      5.5
Bins per symbol:        2048
Samples per symbol:     16384
Decimation:             8
/root/.gnuradio/prefs/vmcircbuf_default_factory: No such file or directory
vmcircbuf_createfilemapping: createfilemapping is not available

(lora_receive_realtime.py:8): Gtk-WARNING **: for_size smaller than min-size (6 < 14) while measuring gadget (node check, owner GtkCheckButton)

(lora_receive_realtime.py:8): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkCheckButton

I was hoping that Docker would make the process of getting started somewhat easier. I'd welcome any further insight you can offer to get this useful tool running.
Thanks

jndqadar, 2022-10-11:
> [root@84dcc5275ba8 apps]# sudo ./lora_receive_file.py
> /usr/lib/python2.7/site-packages/wx-3.0-gtk3/wx/_core.py:16629: UserWarning: wxPython/wxWidgets release number mismatch
> warnings.warn("wxPython/wxWidgets release number mismatch")
> Unable to access the X Display, is $DISPLAY set properly?

I am facing the same error; would you be kind enough to assist me, please?