rpp0/gr-lora #114: PHY Payload CRC implementation

dlz3636, 2020-03-30:
Hello,
I'm interested in adding the CRC check for the PHY payload, a feature which is mentioned in the release notes not to be implemented yet. Some questions..
(1)  Are any of the standard CRC parameters for Lora known (Polynomial, Init Value, XOR, etc) or is the reason it has not been implemented because these are unknown?
(2) In docs/test-results some example payloads are shown with CRC on but the 2 bytes of CRC is not shown in the "pass" result. Are these CRC values available for checking results?
(3) Do the cfiles in the gr-lora-samples repository have corresponding meta data files so they can be run in lora_receive_file_nogui.py ? I was unable to get them to decode correctly by modifying the sample rate and frequency offset values given in the readme to the existing meta data file for the example file.

rpp0, 2020-03-30:
Hi,

This would be great!
1) These parameters have been reverse engineered already by a similar
project: see
https://github.com/myriadrf/LoRa-SDR/blob/52ec64dddf03eefe17cbb36f34377ab097d0abda/LoRaCodes.hpp
at
the function sx1272DataChecksum. The reason this function is not yet in
this repo is because I wanted to find my own methodology of reverse
engineering rather than copying the code. After all, this is a research
project.
2) They are available, but indeed ignored for now. It should be trivial to
add this check to the test suite.
3) Unfortunately no, the metadata in in the filename for these files. They
were transmitted using a RN2483 in explicit header mode.

Snork2, 2021-06-22:
I was able to find this reverse engeneering report at https://www.epfl.ch/labs/tcl/resources-and-sw/lora-phy/

There is a definitive decscription of both CRC algorithms used for header and for payload. Also, there is a code that should be able to perform crc check https://github.com/tapparelj/gr-lora_sdr/blob/4dd27b9a3fed9f97fe840ff9568b0242cbd7d061/lib/crc_verif_impl.cc#L111

I tried to use this algorith to verify CRC value, but I was unable to calculate CRC value that should match 2 last bytes of payload generated by gr-lora. I supose this is because according to cited paper, crc value should not be dewhitened. It as also reflected in code https://github.com/tapparelj/gr-lora_sdr/blob/4dd27b9a3fed9f97fe840ff9568b0242cbd7d061/lib/dewhitening_impl.cc#L101

I supose, that gr-lora is not taking special measures to not dewhitten CRC bytes and this is the reason of failure of my attempt