rpp0/gr-lora #118: Decoding problem at SF11 and SF12

a416485164, 2020-07-03:
I have to say, your work is fantastic. After testing, the results show that the decoding success rate is very high at SF7-SF10 and BW125-BW500. However, the results of SF11-12 and BW125-500 were not satisfactory. Did I set some parameters incorrectly?
Receiver Hardware :  USRP - B200
Transmitter Hardware:RN2483

What is the LoRa Receiver parameter setting for SF11-12 during the test?
Do you have any suggestions?
Thank you again for sharing, and I hope you can take the time to answer my questions. 
Thank you.

rpp0, 2020-07-05:
Thanks for the kind words! Some devices handle SF11-12 differently than others. For example, the RN2483 enables "reduced rate mode" for these SFs whereas the RFM96 doesn't. Therefore, assuming you are receiving in implicit header mode, you could try enabling reduced rate mode.

a416485164, 2020-07-07:
Thank you very much for your help！ I tried the suggestions you said, but the decoding success rate did not increase significantly.
When the transmitter sends the character string “1234” (SF=12, BW=125K), the receiver’s decoding result is  04 9d 40 81 33 33 44 42 fc . The decoding result is the same every time, I think it is because of which setting is wrong, or the decoding process is wrong.  
samp_rate = 1M
decimation = 2
Reduced rate = True
Receiver Hardware : USRP - B200
Transmitter Hardware:RN2483

Transmitter changed to sx1276
When the transmitter sends the character string “1234” (SF=12, BW=125K), the receiver’s decoding result is 04 9d 40 b1 03 03 34 89 88 

samp_rate = 1M
decimation = 2
Reduced rate = False
Receiver Hardware : USRP - B200
Transmitter Hardware:SX1276

Can you give me some suggestions or ideas? Thank you！

a416485164, 2020-08-04:
Regarding the problem of low decoding success rate, I have found the reason. Because the correlation of FIND_SFD is too high, it is okay if it is not so high.  I modified the following code: 
if (c> 0.96f) changed to if (c> 0.95f)
Then，The decoding success rate has increased rapidly.