rpp0/gr-lora #121: error compiling of gr3.7

anonym233, 2020-09-22:
Hello, I have installed the deps and ran the commands to build until make, that fails just here:
[  3%] Building CXX object lib/CMakeFiles/gnuradio-lora.dir/decoder_impl.cc.o
[  6%] Building CXX object lib/CMakeFiles/gnuradio-lora.dir/message_file_sink_impl.cc.o
[  9%] Building CXX object lib/CMakeFiles/gnuradio-lora.dir/message_socket_sink_impl.cc.o
[ 12%] Building CXX object lib/CMakeFiles/gnuradio-lora.dir/channelizer_impl.cc.o
[ 16%] Building CXX object lib/CMakeFiles/gnuradio-lora.dir/controller_impl.cc.o
[ 19%] Building CXX object lib/CMakeFiles/gnuradio-lora.dir/debugger.cc.o
[ 22%] Building CXX object lib/CMakeFiles/gnuradio-lora.dir/message_socket_source_impl.cc.o
[ 25%] Linking CXX shared library libgnuradio-lora.so
[ 25%] Built target gnuradio-lora
[ 29%] Building CXX object lib/CMakeFiles/test-lora.dir/test_lora.cc.o
/home/xhq/grlora/gr-lora/lib/test_lora.cc:36:11: warning: unused parameter ‘argc’ [-Wunused-parameter]
 main (int argc, char **argv)
           ^
/home/xhq/grlora/gr-lora/lib/test_lora.cc:36:24: warning: unused parameter ‘argv’ [-Wunused-parameter]
 main (int argc, char **argv)
                        ^
[ 32%] Building CXX object lib/CMakeFiles/test-lora.dir/qa_lora.cc.o
[ 35%] Building CXX object lib/CMakeFiles/test-lora.dir/qa_message_socket_sink.cc.o
[ 38%] Linking CXX executable test-lora
CMakeFiles/test-lora.dir/qa_message_socket_sink.cc.o：in function ‘_GLOBAL__sub_I__ZN2gr4lora22qa_message_socket_sink2t1Ev’：
qa_message_socket_sink.cc:(.text.startup+0x38)：undefined reference to ‘log4cpp::Appender::AppenderMapStorageInitializer::AppenderMapStorageInitializer()’
qa_message_socket_sink.cc:(.text.startup+0x47)：undefined reference to‘log4cpp::Appender::AppenderMapStorageInitializer::~AppenderMapStorageInitializer()’
collect2: error: ld returned 1 exit status
lib/CMakeFiles/test-lora.dir/build.make:125: recipe for target 'lib/test-lora' failed
make[2]: *** [lib/test-lora] Error 1
CMakeFiles/Makefile2:179: recipe for target 'lib/CMakeFiles/test-lora.dir/all' failed
make[1]: *** [lib/CMakeFiles/test-lora.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

ubuntu 16.04
gnuradio is 3.7.10
log4cpp is 1.1.3
gcc is 5.4.0
cmake is 3.15.7

riatankarsahu, 2020-11-19:
Hello @anonym233 
I am not very sure, but probably that repository had been upgraded and the present requirement will be gnuradio - 3.8.x.
Hence Ubuntu 16.04 can not be a good choice for it as it is limited to gnuradio 3.7.x. 
I am using following configuration and I could successfully install everything. But I am now having new problem. Still if it's only about installation then you can use undermentioned configuration.

ubuntu - 20.04
gnuradio - 3.8.x
python - 3.x
Good Luck!!

ph03n1xAu, 2021-03-01:
Hi @anonym233,

I Installed the gr-lora library of gr3.7 branch today in ubuntu 16.04. Its working perfectly in my system. 

If you are still having problem, I would suggest to check the dependencies: python2-numpy, python2-scipy, swig, cppunit, fftw, gnuradio, libvolk, log4cpp, cmake, wx, and liquid-dsp. 

Cheers!!