rpp0/gr-lora #125: ModuleNotFoundError: No module named 'lora'

riatankarsahu, 2020-11-19:
Hello @rpp0 
I am experiencing a problem while I am working with this repository. I installed everything manually according to the 
[image]
build passing tab information. But now I am having problem while I am trying to decode a message from file supplied by you. It's saying 
import lora
ModuleNotFoundError: No module named 'lora'
I don't know what did I do wrong! Can you please help me? I will be grateful to you!!!
[image]

zahraaghapour, 2020-11-19:
Hello @riatankarsahu
I had this problem(Ubuntu), I added, "`usr/local/lib/python2.7/site-packages`" to `$PYTHONPATH` then it was solved.
with this command:` export PYTHONPATH=$PYTHONPATH:usr/local/lib/python2.7/site-packages` then you can check if it's added correctly to pythonpath with this command `printenv`.
have a look at this [link](https://www.epfl.ch/labs/tcl/resources-and-sw/lora-phy/) . it's mentioned at the end of the page as frequent errors.

ph03n1xAu, 2020-11-24:
I'm also experiencing this problem
```
Traceback (most recent call last):
  File "/home/sdr/Documents/gnuradio/lora_receive_realtime.py", line 34, in <module>
    import lora
**ModuleNotFoundError: No module named 'lora'**

>>> Done (return code 1)
```
export PYTHONPATH didn't resolve the error. I'm running gnuradio 3.8 on Ubuntu 20.04.1 LTS. 
I tried to import the module lora from python IDE, still same problem..
looks like something else is missing.. unable to figure it out yet

riatankarsahu, 2020-11-24:
Hello phoenix(@ph03n1xAu ),
Since Ubuntu 17.04 that solution has been obsoleted. Because the python update which is for Ubuntu 20.04 probably python 3.6. Can you please send me the screenshot of that full message generated by your flowgraph? I need to see one thing. I think I have the solution. Thank you in advance.
-- Ritam

riatankarsahu, 2020-11-24:
Hi Zahra (@zahraaghapour ),
Thank you so much for the solution! Though it didn't fix the problem but I found the solution through that way. It's a problem of python version swapping as now a days python 3.x is in use so site-packages are no longer in use. I had to change the PYTHONPATH to 3.6 dist-package. If you wish to upgrade ubuntu you can think about it. Thank you once again!!!
--Ritam

zahraaghapour, 2020-11-24:
Hi Ritankar @riatankarsahu 
I am glad you solved your problem.
Good luck :)

ph03n1xAu, 2020-11-24:
> Hello phoenix(@ph03n1xAu ),
> Since Ubuntu 17.04 that solution has been obsoleted. Because the python update which is for Ubuntu 20.04 probably python 3.6. Can you please send me the screenshot of [...]

Hi Ritam, @riatankarsahu 

I found that in my vm python3 is linked to python3.8
`lrwxrwxrwx 1 root root 9 Mar 13  2020 /usr/bin/python3 -> python3.8`

And this is where the rpp0/gr-lora installed all the files by default:

```
sdr@LimeLab:/usr/local/lib/python3/dist-packages/lora$ ls
__init__.py   __init__.pyo   loraconfig.pyc  lora_receiver.py   lora_receiver.pyo  lorasocket.pyc
__init__.pyc  loraconfig.py  loraconfig.pyo  lora_receiver.pyc  lorasocket.py      lorasocket.pyo
```

I tried additional few things today:
-- copied all lora *.py files under python3.8 dist-packages..
-- set PYTHONPATH to 3.8 dist-package..didn't work
I guess I'm missing something obvious :-(

Here is the screenshot for the gnuradio error:
[image]

Appreciate any suggestions..

riatankarsahu, 2020-11-25:
Hello Phoenix(@ph03n1xAu ),
Please look at the picture :D 
1. Gnu Radio is looking for the module in /usr/bin/python3 (not /usr/bin/python3.8)
2. Please look at the usr/local/lib/python3/dist-packages (if the folder lora exist there).
3. If yes then restart the pc.
4. Execute the command: export PYTHONPATH=$PYTHONPATH:usr/local/lib/python3/dist-packages
5. Execute the command: printenv
6. Run Gnu Radio and try to activate the flowgraph.
If it doesn't work then need to go through hard way. And if you have done it, please notify me. Then there is other ways.
Good Luck!!
-- Ritam

riatankarsahu, 2020-11-25:
> Hi Ritankar @riatankarsahu
> I am glad you solved your problem.
> Good luck :)

Good Luck to you too!! :D

ph03n1xAu, 2020-11-25:
Hi Ritam @riatankarsahu ,

I noticed gnuradio is looking for python3 (/usr/bin/python3 -u /home/sdr/Documents/gnuradio/lora_receive_realtime.py) but this /usr/bin/python3 is actually a soft-link for python3.8

[image]

Also I exported the pythonpath previously..did fresh restart and set again just now..You can see the lora files under dist-package
still same issues..ModuleNotFoundError

[image]

riatankarsahu, 2020-11-25:
> Hi Ritam @riatankarsahu ,
> 
> Thanks for your suggestions..yes, I noticed gnuradio is looking for python3 (/usr/bin/python3 -u /home/sdr/Documents/gnuradio/lora_receive_realtime.py) but this /usr/bin/python3 is actually a soft-link for python3.8
> 
> [image]
> 
> Also I exported the pythonpath previously..did fresh restart and set again just now..You can see the lora files under dist-package
> still same issues..ModuleNotFoundError
> 
> [image]

Hey Pho

riatankarsahu, 2020-11-25:
Hey @ph03n1xAu ,
Did you replace the whole dist-package from python 3 to 3.8? Not only the files but the whole folder. Or only the lora folder from python3/dist-packages to python3.8/dist-packages. Please try that once. Good luck!!
-- Ritam

ph03n1xAu, 2020-11-25:
yes I did that already..Unfortunately, problem still persists..
i'll try few other things..let's see..will let you know..
Have a good night :-)

ph03n1xAu, 2020-11-26:
Hi Ritam @riatankarsahu ,
I reckon, I found the problem but not sure is it the only thing that I've to fix now..I re-built gr-lora today. I found in cmake steps swig installation had some issues

[image]

would you please share the contents of 'gr-lora/build/install_manifest.txt' files. This file is probably created after running cmake command. You should have some entry for lora_swig.py which is missing in my setup.

riatankarsahu, 2020-11-26:
Hi Phoenix @ph03n1xAu ,
I am sending the content below. Have a look. And also I doubt about one thing. Could you please tell me that how did you install the Gnu Radio? I mean from which link? That looks like dependency mismatch....! Good Luck! 
[install_manifest.txt](https://github.com/rpp0/gr-lora/files/5603128/install_manifest.txt)

-- Ritam

ph03n1xAu, 2020-11-27:
Hi Ritam @riatankarsahu ,

Thanks for sharing the file contents. 
Yesterday, after installing new version SWIG library the problem is resolved. Now, the error ('no module named lora') is not there anymore and it's running successfully.

However, have you got any success on decoding? I tried to decode message ('hello world') from Arduino-LoRa shield (HopeRF95 chip) yesterday. it didn't work. I'll try with few other parameters today.

Hope to be in contact with you for this lora project with limeSDR :-)

riatankarsahu, 2020-11-27:
Hi Phoenix @ph03n1xAu 
Congrats on your success!! I can prepare or collect a few *.iq file for you by which you can simulate your model after creating the flowgraph. I finished my project on it and yes it works perfectly till an extent. Rest of the part I will try to implement after Monday. I am closing that thread now. If you wish you can contact me personally. Have a nice day!!! :D 
-- Ritam

closed 2020-11-27 by riatankarsahu

menna45, 2021-06-23:
@riatankarsahu @ph03n1xAu  i have the same error i downloaded GNURadio 3.8 from source ppa installation for ubunto 20.04 , downloaded gr-lora from @rpp0  and i have read all you comments and did what you said in the comments but the error is still 
[image]
[image]
[image]
[image]
[image]