rpp0/gr-lora #126: Docker Image and Manual Install Issues

epgreen, 2020-11-24:
I'm getting a "No Command Specified" error when I try to run the docker image. 

And when I do the manual install I'm getting an error:

Traceback (most recent call last):
  File "./lora_receive_file_nogui.py", line 11, in <module>
    from lora.loraconfig import LoRaConfig
ModuleNotFoundError: No module named 'lora.loraconfig'

Any help would be appreciated, I see the issue has come up before but I didn't see any solution that has worked for me.

epgreen, 2020-11-24:
I'm still getting the docker issue of not having a command. I believe that one is due to a missing CMD at the end, but I'm not familiar enough with docker to know how to find that out or change it.

With the manual install and the "ModuleNotFoundError" I got a solution:

I'm using a vm called 'instant-gnuradio.ova' which has been handy. Some issues could have been due to configuration choices since it's a pre-built VM. 

PYTHONPATH=/home/gnuradio/.local/lib/python2.7/site-packages

The issue is that the "lora" module is saved by default in /usr/local/lib/python3/dist-packages and the one that was saved in /home/gnuradio/.local/lib/python2.7/site-packages doesn't work I suppose. I renamed the old one to lora_old, and moved the 'lora' module from the python3/dist-packages folder to the python2.7/site-packages folder. 

With that, on line 67 of the lora_receive_file_nogui.py it calls for "raw_input" that was erroring out, I replaced that with "input" and it worked just fine. 

This may cause issues later, but wanted to post in case someone runs into a similar issue.

riatankarsahu, 2021-01-02:
Hello @epgreen 
Please follow our discussion on that thread : https://github.com/rpp0/gr-lora/issues/125

I think you will find something useful. Good Luck!!!
-- Ritam