rpp0/gr-lora #129: Missing LoRa module

MartinPolytech, 2021-01-27:
Hello everyone :)

We are currently working on a lora project and we are having some issues with the installation of gr-lora.  After finishing the installation the lora blocks seem to be missing (as you can see in the lora_receive_file below).

[image]

And when we try to execute the file "lora_receive_file_nogui.py" file to test the installation, the terminal says that the lora modules are missing.

[image]

We have checked at usr/local/lib/python3/dist-packages and the lora folder is missing, could you tell us how to fix this ?

Any help is appreciated :)

rpp0, 2021-01-27:
Hello, during the execution of `sudo make install` you should see the directory to which the files are copied. My guess is that they are not copied to `/usr/local/lib`. You can specify the install directory with `-DCMAKE_INSTALL_PREFIX=` as indicated in the installation instructions.

MartinPolytech, 2021-01-28:
Hello,

First of all thank you for the quick answer :) With your advice we managed to install gr-lora correctly.

Howewer when we try to execute the file "lora_receive_file_nogui.py" to test the installation, the following error appears. 

[image]

Is it and error from us or in the file ?

Thank you again and have a great day !

rpp0, 2021-01-30:
A while ago, gr-lora transitioned to Python 3. I think I forgot to update this example file. It should work if you use `input` rather than `raw_input`. I'll post a patch this weekend, thanks for reporting!

rpp0, 2021-02-01:
Fixed in 124f2120ae29583a4ca2d2a5e6cb9eb67c2d1b4e!

closed 2021-02-01 by rpp0