rpp0/gr-lora #132: LimeSDR and LoPy4 - Can't decode LoRa signal

msalinascamus, 2021-03-03:
Hi
I'm capturing and decoding a LoRa signal (I'm sending the same packet with a LoPy4 every 6 seconds) using a LimeSDR. But I'm not able to decode the LoRa packets.
This is the flowgraph that I'm executing:
[image]
And this is what I get in the console:
[image]
I just get random or weird symbols when decoding, I been also using this repository https://github.com/PentHertz/LoRa_Craft but it hasn't help me.
I thought that it could be the decimation or maybe the channel that I'm using but still haven't got good results.
Thanks for reading, have a good day!

rpp0, 2021-03-04:
Hello

Can you send me a compressed raw capture file? I'd be interested to look at the signal.

msalinascamus, 2021-03-05:
This is what I get in the file sink
[image]

And this is what I get in the frequency waterfall plot
[image]

rpp0, 2021-03-05:
I can't help you with that information alone; I need a capture file. That said, I can already see two potential misconfigurations:
- The channel list is set at 915.5 MHz but your signal seems to be around 914.9 MHz.
- Based on eyeballing the frequency plot it seems your bandwidth is closer to 250 KHz instead of 125 KHz.

It's important that the parameters you set in gr-lora are the same values your transmitter uses.

msalinascamus, 2021-03-08:
- **The channel list is set at 915.5 MHz but your signal seems to be around 914.9 MHz.** I'm sorry about this, I took the screenshot when I was testing other possible channels. But I make sure every time that I'm receiving and transmitting in the same frequency.

- **Based on eyeballing the frequency plot it seems your bandwidth is closer to 250 KHz instead of 125 KHz** I'm using SF7BW125 in the config file to transmit using LoPy4.

- The capture file can be downloaded here.[https://drive.google.com/file/d/1_divjfj6yGVe5KUl7QtyStGVEEnSabz6/view?usp=sharing](url)

rpp0, 2021-03-08:
Your SDR appears to be configured at the wrong center frequency. The LoRa signal is cut off at the edges of the bandwidth.

[image]

[image]

This is probably why the signal doesn't decode properly.

msalinascamus, 2021-03-08:
Thanks for the answer, but it seems weird to me that the SDR is configured at another center frequency.
This picture shows the parameters of the LoRa Receiver block
[image]
And the last two screenshot show the parameters of the LimeSDR
[image]
[image]

freq_slider is set on default at 915MHz, gain at 50dB and samp_rate at 1MHz

rpp0, 2021-03-09:
What I mean with wrong center frequency is: in your capture file you appear to be transmitting a LoRa message at 915.5 MHz, while your SDR is tuned to 915 MHz with a bandwidth of 1 MHz. This means that 62,500 Hz from your LoRa signal will be cut off (as shown in my screenshot). 

So, to fix your problem, you can either configure your SDR's center frequency so that the full signal is within the bandwidth of 1 MHz, or increase the bandwidth, or transmit at a channel that is closer to your center frequency using the LoPy. Please refer to the tutorial if anything is still unclear: https://github.com/rpp0/gr-lora/wiki/Capturing-LoRa-signals-using-an-RTL-SDR-device