rpp0/gr-lora #133: About the FFT part in decoder implementation

qinchuanhui, 2021-03-07:
I notice that the FFT method is implemented in the code, such as the function:  `decoder_impl::get_shift_fft`. However, in the demodulation part the usage of the fft method is commented. As the FFT is the basic method in Lora, I wonder why you refuse to use it while already finishing writing it. Are there some errors when using FFT?

rpp0, 2021-03-07:
The FFT method is indeed implemented, but currently only works well in gr-lora when using ideal signals (that is, signals without frequency or time offset errors).  At the time, there was few / no documentation on how the LoRa frequency / time synchronization works when applying the FFT method, so I didn't understand properly how it was done in the real hardware. However, in the meantime I saw that a few papers came out with descriptions of a synchronization algorithm, and given the recent surge in interest in gr-lora, I might revisit the problem soon to implement the FFT method fully.

qinchuanhui, 2021-03-08:
Thanks a lot for your answer, I will also try to find the implementation of FFT.

closed 2021-03-14 by qinchuanhui