rpp0/gr-lora #138: Where to download the Lora specification?

commthu, 2021-05-24:
Can you give me the link to the specifications relate your code.
Thank you very much.

CuBiC3D, 2021-06-20:
LoRa is still proprietary

Amalinda, 2021-06-21:
You might find the Pieter Robyns ( maintainer of this repo) publication titled "gr-lora insightful A Multi-Channel Software Decoder for the LoRa Modulation Scheme" through https://pdfs.semanticscholar.org/2907/d79a220965ad64f4ce9cfcda2f291ecd19bc.pdf

Amalinda, 2021-06-21:
You can find Pieter Robyns (maintainer of this repo) publication titled "gr-lora insightful A Multi-Channel Software Decoder for the LoRa Modulation Scheme" through https://pdfs.semanticscholar.org/2907/d79a220965ad64f4ce9cfcda2f291ecd19bc.pdf