rpp0/gr-lora #151: Decoding problem with RMF95W

fmc5046, 2022-01-10:
Hi I'm having issues decoding a message transmitted by a ESP32 transmitting from a RMF95W LoRa Transceiver.

I'm transmitting with the following settings:
SF: 7, BW: 125k, C/R: 4/8, Frequency: 915MHz (Bellow is the screenshot of the Arduino IDE configuring the transmit settings.  I am transmitting a simple string of "AABBCC"

[image]

On the GNU radio side I have the following setup (I am using an RTL-SDR).  Based on the tutorial I implemented an offset:

[image]

Based on info from other closed issues I used Explicit headers (Implicit header false) and I can see what I judge to be a reasonable LoRa packet:

[image]

Here is the decoded output:

[image]

Any ideas what may be the issue?  Thanks for the help!

jndqadar, 2022-10-20:
Hi, Can you please help me how can I find "LoRa Receiver and Message socker sink" blocks in my GNU?

jndqadar, 2022-10-20:
Can you please upload a short video tutorial on how to do this process?