rpp0/gr-lora #153: Can't set channel or see any packet output with `apps/lora_receive_realtime.py` in the Docker container

interfect, 2022-02-01:
The recommended Docker-based way of installing and using gr-lora is to run `docker/docker_run_grlora.sh`, which then drops you into a Docker container.

The Docker container has the `lora_receive_realtime.py`, and I can run it and have it pick up my RTL-SDR and show an FFT. If I tune my transmitter to 868 MHz, spreading factor 2048 chips/symb (AKA 11), and bandwidth 125000, and transmit, I can see a signal pop up in the FFT plot.

But:

1. I don't see any decoded output being printed. The README says that decoded output would be logged to the console by default, but the `lora_receive_realtime.py` script sets up a `lora.message_socket_sink`, so I think if it's really going anywhere it's going out over UDP. But it's going to 127.0.0.1 port 40868, and because the Docker isn't run with `--net=host` I think it's just going to the Docker container itself, where I can't really get at it.
2. The Docker container doesn't ship a text editor. So I can't edit `lora_receive_realtime.py` to send the output to a different IP or port, or to print it instead of sending it to a socket. **I also cannot tune the receiver to the channel, spreading factor, etc. that I actually want to receive on.**
3. (Minor point) What coding rate is the receiver's default? It isn't set in `lora_receive_realtime.py`, so I'm not certain that my test packets are being sent at the right coding rate, so I'm not certain whether I should be expecting to see any decoded output.

Am I actually supposed to be trying to use the Docker container as a way to actually receive LoRa data? Or is it just a demo?

raul-parada, 2023-01-10:
Hi @interfect, did you make it work? I also don't get the decode packet. I get only this information while running.
[image]

And this plot
 
[image]

thanks

interfect, 2023-01-11:
I never managed to make it work either, unfortunately.