rpp0/gr-lora #179: Python 2-shebang in `lora_receiver.py` -- is this a Python 3 or Python 2 software?

dreirund, 2024-05-08:
Although this software seems to build fine for python 3.12, one of the built python files still has Python 2 in it's shebang line:

[`lora_receiver.py`](https://github.com/rpp0/gr-lora/blob/master/python/lora_receiver.py) starts with  
´´´python
#!/usr/bin/env python2
```

Is this software fully python3-supported and this was only forgotten? (Then remove the `2`.) 
Or does this still depend on Python 2?

Regards!

eabase, 2024-11-06:
good catch

dreirund, 2024-11-06:
> *good catch, there are several.*

@rpp0, and what is the official state of the issue?

*(If you do not maintain this anymore, can you leave a note please in the [`README.md`](https://github.com/rpp0/gr-lora/blob/master/README.md) and archive this repository?)*

Regards and thanks for maintaining!